MiCode/Xiaomi_Kernel_OpenSource #2349: SEVERAL CAMERA, DRM AND TOUCHSCREEN DRIVERS ARE MISSING ON CAMELLIA-R-OSS (POCO M3 PRO 5G/REDMI NOTE 10 5G/REDMI NOTE 10T 5G)

MTK0DIMENSITY0GANG, 2021-12-07:
Hello, I noticied on stock MIUI defconfig that focaltech directory driver is FT8722, however that driver doesn't exist on camellia-r-oss which mean that touchscreen not gonna work on unit who use a focaltech touchscreen. Could you please add that driver ?

**EDIT:
ov02b10_aac_macro_mipi_raw camera driver is missing
ft8722_fhdp_dsi_vdo_tianma_90hz_k19 and nt36672c_fhdp_dsi_vdo_truly2_90hz_k19 lcm drivers are missing
NT36672C TRULY2 and FT8722 TIANMA drm drivers are missing**

#2270 #2326 [CAMELLIA-R-OSS](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/camellia-r-oss)

MTK0DIMENSITY0GANG, 2021-12-07:
@mi-code @liuguo09

xdferchoz14, 2022-06-03:
@mi-code @liuguo09 please help, it would be great help in software development