MiCode/Xiaomi_Kernel_OpenSource #2351: cust.dtsi missing POCO M3 Pro camellia-r-oss

Kiolli, 2021-12-08:
DTC     arch/arm64/boot/dts/mediatek/camellia.dtb
../arch/arm64/boot/dts/mediatek/camellia.dts:457:10: fatal error: 'camellia/cust.dtsi' file not found
#include "camellia/cust.dtsi"                                                  ^~~~~~~~~~~~~~~~~~~~

willzyx-hub, 2021-12-08:
> DTC arch/arm64/boot/dts/mediatek/camellia.dtb ../arch/arm64/boot/dts/mediatek/camellia.dts:457:10: fatal error: 'camellia/cust.dtsi' file not found #include "camellia/cust.dtsi" ^~~~~~~~~~~~~~~~~~~~

Yes that's file is missing from source too... Since this is open source why can't you complete it ? @mi-code

MTK0DIMENSITY0GANG, 2021-12-08:
@mi-code

Kiolli, 2021-12-09:
@mi-code

androiddisk, 2023-09-21:
@mi-code

AbzRaider, 2024-02-21:
you should generate that using dct tools anyways there is missing lcm drivers so no use