MiCode/Xiaomi_Kernel_OpenSource #2355: MI Note 3 jason-n-oss 编译报错

suqiernb, 2021-12-25:
* 按照 How To Use 所说, 我下载了 QAEP
* 在编译过程中我出现了我看不懂的错误
```bash
# ......
[ 14% 359/2559] target asm: libc <= bionic/libc/arch-common/bionic/crtbrand.S
FAILED: /bin/bash -c "(PWD=/proc/self/cwd /clang  -Wno-vectorizer-no-neon -I external/jemalloc/include -I bionic/libc/stdio -I bionic/libc -I out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates -I out/target/product/sdm660_64/gen/SHARED_LIBRARIES/libc_intermediates -I libnativehelper/include/nativehelper \$(cat out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/import_includes) -isystem system/core/include -isystem system/media/audio/include -isystem hardware/libhardware/include -isystem hardware/libhardware_legacy/include -isystem hardware/ril/include -isystem libnativehelper/include -isystem frameworks/native/include -isystem frameworks/native/opengl/include -isystem frameworks/av/include -isystem frameworks/base/include -isystem out/target/product/sdm660_64/obj/include -isystem bionic/libc/arch-arm64/include -isystem bionic/libc/include -isystem bionic/libc/kernel/uapi -isystem bionic/libc/kernel/common -isystem bionic/libc/kernel/uapi/asm-arm64 -isystem bionic/libm/include -isystem bionic/libm/include/arm64 -c    -fno-exceptions -Wno-multichar -fno-strict-aliasing -fstack-protector-strong -ffunction-sections -fdata-sections -funwind-tables -Wa,--noexecstack -Werror=format-security -D_FORTIFY_SOURCE=2 -fno-short-enums -no-canonical-prefixes -Werror=pointer-to-int-cast -Werror=int-to-pointer-cast -Werror=implicit-function-declaration -DANDROID -fmessage-length=0 -W -Wall -Wno-unused -Winit-self -Wpointer-arith -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Werror=date-time -DNDEBUG -O2 -g -Wstrict-aliasing=2 -DNDEBUG -UDEBUG  -D__compiler_offsetof=__builtin_offsetof -Werror=int-conversion -Wno-reserved-id-macro -Wno-format-pedantic -Wno-unused-command-line-argument -fcolor-diagnostics -nostdlibinc     -target aarch64-linux-android -Bprebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/aarch64-linux-android/bin    -std=gnu99        -Ibionic/libc/include -DPLATFORM_SDK_VERSION=25 -D__ASSEMBLY__ -MD -MF out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.d -o out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.o bionic/libc/arch-common/bionic/crtbrand.S ) && (cp out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.d out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.P; sed -e 's/#.*//' -e 's/^[^:]*: *//' -e 's/ *\\\\\$//' -e '/^\$/ d' -e 's/\$/ :/' < out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.d >> out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.P; rm -f out/target/product/sdm660_64/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.d )"
/bin/bash: /clang: No such file or directory
[ 14% 359/2559] build out/target/product/sdm660_64/obj/lib/crtbegin_so1.o
ninja: build stopped: subcommand failed.
build/core/ninja.mk:148: recipe for target 'ninja_wrapper' failed
make: *** [ninja_wrapper] Error 1

#### make failed to build some targets (47 seconds) ####
```

suqiernb, 2021-12-25:
求助, 我不知道是我哪里做错了, 目前我无法解决这个问题

Aquarius223, 2021-12-25:
你可以按照此地方的教程来编译内核: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone
* 参考Max3部分即可

suqiernb, 2021-12-25:
> 你可以按照这个地方的教程来编译内核：[https](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone) : [//github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone)
> 
> * 参考Max3部分分类

独立编译内核的话没有 wlan

Angel0726, 2022-02-01:
> 独立编译内核的话没有 wlan

独立编译内核没有wlan?是没有WIFI吗？