MiCode/Xiaomi_Kernel_OpenSource #2356: fastboot boot command is bugged on Xiaomi MTK devices

pjgowtham, 2021-12-27:
The best practice for installing TWRP seems to be to inject ramdisk after using fastboot boot twrp.img command. But when this command is broken on the firmware side, I have to make new TWRP build everytime you push OTA cos the wifi/BT/etc kernel modules wont load if the TWRP kernel (latest stock kernel at the time of building) does not match cos devices nowadays use recovery as boot.

It sometimes says "successful" but ends up booting into system rather than TWRP even when twrpfastboot=1 cmdline (forces to boot twrp) is on and randomly throws error: Too many links even with using usb 2 hub.

It would make life a lot easier if you could fix this on your MTK lineup devices. If you cant fix it, release devices with dedicated recovery partition, Thanks.

willzyx-hub, 2021-12-29:
> The best practice for installing TWRP seems to be to inject ramdisk after using fastboot boot twrp.img command. But when this command is broken on the firmware side, I have to make new TWRP build [...]

Hello, i agree with you this is MTK bootloader failure (lk.img) the boot command doesn't seem to work and it's return an error 

$fastboot boot (name of file).img
Sending (name of file).img -> OKAY
Booting... -> FAILED : (Status read failed (No such device) ) and ofc @mi-code you're From Xiaomi Team pls make it work by patching lk.img because only you can build and sign the lk.img Thanks :)

rama982, 2021-12-29:
They will never do that tho.
Its happpen in all mediatek devices not only Xiaomi, but other also.
The solution is simple just do fastboot flash boot then reboot to twrp -> restore stock boot.img and do ramdisk patching.
No offense.

lucaszanella, 2022-02-28:
On my Poco M3:
```

fastboot boot Image
creating boot image...
creating boot image - 35883008 bytes
Sending 'boot.img' (35042 KB)                      OKAY [  0.837s]
Booting                                            FAILED (remote: 'unknown command')
fastboot: error: Command failed
```

Sushrut1101, 2022-04-24:
please fix it asap

railjty, 2022-07-12:
It seems that Mediatek does not want to add `fastboot boot`

XxMarcelo, 2022-08-04:
tente esses comandos :
fastboot --disable-verity --disable-verification flash vbmeta vbmeta.img
fastboot flash boot <arquivo twrp>
fastboot reboot recovery
E se não funcionar é só baixar uma ROM fastboot atual do seu dispositivo e extrair o arquivo boot. img e fazer esse comando: fastboot flash boot boot.img 
O arquivo vbmeta.img também é encontrado quando extrai a ROM fastboot, por favor avise se funcionar..

kde-yyds, 2025-03-19:
> On my Poco M3:
> 
> ```
> 
> fastboot boot Image
> creating boot image...
> creating boot image - 35883008 bytes
> Sending 'boot.img' (35042 KB)                      OKAY [  0.837s]
> Booting                                            FAILED (remote: 'unknown command')
> fastboot: error: Command failed
> ```

Same on xaga