MiCode/Xiaomi_Kernel_OpenSource #2357: Camellia kernel source is broken

pjgowtham, 2021-12-27:
As usual cust.dtsi is missing but its unlikely that its the issue

You have 3 touchpanel drivers for the same touchpanel ( wherein two of them starts building automatically - one hardcoded and one built cos of defconfig). The third one being unmodified novatek driver. None of this fixes the kernel and it end up either bricking or bootlooping.

There is also missing drivers for certain touchpanels and sensors. Here's a diff between your stock camellia defconfig and the generated defconfig. Kindly sync the source with your working tree or even better release camellia-s-oss source early.

203c203
< CONFIG_KALLSYMS_ALL=y
---
> # CONFIG_KALLSYMS_ALL is not set
222a223
> # CONFIG_MIHW is not set
281c282
< # CONFIG_LTO is not set
---
> CONFIG_LTO=y
284,285c285,291
< CONFIG_LTO_NONE=y
< # CONFIG_LTO_CLANG is not set
---
> CONFIG_THINLTO=y
> # CONFIG_LTO_NONE is not set
> CONFIG_LTO_CLANG=y
> CONFIG_CFI=y
> # CONFIG_CFI_PERMISSIVE is not set
> CONFIG_CFI_CLANG=y
> CONFIG_CFI_CLANG_SHADOW=y
315c321
< CONFIG_REFCOUNT_FULL=y
---
> # CONFIG_REFCOUNT_FULL is not set
578d583
< CONFIG_RANDOMIZE_MODULE_REGION_FULL=y
588d592
< CONFIG_BUILD_ARM64_DTB_OVERLAY_IMAGE=y
594d597
< CONFIG_BUILD_ARM64_DTB_OVERLAY_IMAGE_NAMES="mediatek/camellia"
1051c1054
< CONFIG_NET_SCH_NETEM=y
---
> # CONFIG_NET_SCH_NETEM is not set
1233d1235
< CONFIG_OF_MDIO=y
1240c1242,1243
< # CONFIG_ZRAM_WRITEBACK is not set
---
> CONFIG_ZRAM_WRITEBACK=y
> CONFIG_MIUI_ZRAM_MEMORY_TRACKING=y
1249c1252
< CONFIG_BLK_DEV_RAM_SIZE=8192
---
> CONFIG_BLK_DEV_RAM_SIZE=4096
1442c1445
< CONFIG_CUSTOM_KERNEL_IMGSENSOR="ov48b_qtech_main_mipi_raw ov48b_aac_main_mipi_raw ov8856_ofilm_front_mipi_raw ov8856_aac_front_mipi_raw gc02m1b_qtech_depth_mipi_mono gc02m1b_aac_depth_mipi_mono hi259h_qtech_macro_mipi_raw hi259h_aac_macro_mipi_raw"
---
> CONFIG_CUSTOM_KERNEL_IMGSENSOR="ov48b_qtech_main_mipi_raw ov48b_aac_main_mipi_raw ov8856_ofilm_front_mipi_raw ov8856_aac_front_mipi_raw gc02m1b_qtech_depth_mipi_mono gc02m1b_aac_depth_mipi_mono hi259h_qtech_macro_mipi_raw hi259h_aac_macro_mipi_raw ov02b10_aac_macro_mipi_raw"
1466c1469
< CONFIG_CUSTOM_KERNEL_LCM="nt36672A_fhdp_dsi_vdo_tianma_j19 nt36672c_fhdp_dsi_vdo_tianma_90hz_k19 nt36672d_fhdp_dsi_vdo_tianma_90hz_k19 nt36672c_fhdp_dsi_vdo_truly_90hz_k19"
---
> CONFIG_CUSTOM_KERNEL_LCM="nt36672A_fhdp_dsi_vdo_tianma_j19 nt36672c_fhdp_dsi_vdo_tianma_90hz_k19 nt36672d_fhdp_dsi_vdo_tianma_90hz_k19 nt36672c_fhdp_dsi_vdo_truly_90hz_k19  ft8722_fhdp_dsi_vdo_tianma_90hz_k19 nt36672c_fhdp_dsi_vdo_truly2_90hz_k19"
1953c1956
< CONFIG_MTK_VOW_DUAL_MIC_SUPPORT=y
---
> # CONFIG_MTK_VOW_DUAL_MIC_SUPPORT is not set
1955,1957c1958,1960
< # CONFIG_MTK_VOW_AMAZON_SUPPORT is not set
< # CONFIG_MTK_VOW_GVA_SUPPORT is not set
< CONFIG_MTK_VOW_MAX_PDK_NUMBER=1
---
> CONFIG_MTK_VOW_AMAZON_SUPPORT=y
[... 254 lines cut ...]
< CONFIG_USB_ANNOUNCE_NEW_DEVICES=y
---
> # CONFIG_USB_ANNOUNCE_NEW_DEVICES is not set
3677,3680c3597
< CONFIG_USB_EHCI_HCD=y
< # CONFIG_USB_EHCI_ROOT_HUB_TT is not set
< CONFIG_USB_EHCI_TT_NEWSCHED=y
< # CONFIG_USB_EHCI_HCD_PLATFORM is not set
---
> # CONFIG_USB_EHCI_HCD is not set
3755d3671
< # CONFIG_USB_SISUSBVGA is not set
4203a4120,4121
> # CONFIG_RTMM is not set
> # CONFIG_KTRACE is not set
4899,4900c4817,4818
< CONFIG_TMPFS_POSIX_ACL=y
< CONFIG_TMPFS_XATTR=y
---
> # CONFIG_TMPFS_POSIX_ACL is not set
> # CONFIG_TMPFS_XATTR is not set
5067c4985
< CONFIG_PANIC_TIMEOUT=5
---
> CONFIG_PANIC_TIMEOUT=1

pjgowtham, 2021-12-27:
`df`

pjgowtham, 2021-12-27:
[diff_stock_generated.txt](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/files/7779087/diff_stock_generated.txt)

z-nerd, 2024-03-26:
DTC     arch/arm64/boot/dts/mediatek/camellia.dtb
    9   ../arch/arm64/boot/dts/mediatek/camellia.dts:457:10: fatal error: 'camellia/cust.dtsi' file not found
   10   #include <camellia/cust.dtsi>
   11            ^~~~~~~~~~~~~~~~~~~~
   12   1 error generated.
   13   make[3]: *** [scripts/Makefile.lib:321: arch/arm64/boot/dts/mediatek/camellia.dtb] Error 1
   14   make[2]: *** [../scripts/Makefile.build:671: arch/arm64/boot/dts/mediatek] Error 2
   15   make[1]: *** [arch/arm64/Makefile:194: dtbs] Error 2

I can't compile kernel with above error because of cust.dtsi didn't exit is source code!