MiCode/Xiaomi_Kernel_OpenSource #2360: recipe for target 'drivers/misc/mediatek' failed

jonjn, 2021-12-30:
Hi, dear Xiaomi developers, when I try to build the kernel branch `bomb-q-oss` (for Redmi 10X and Redmi 10X Pro Android Q) by `aarch64-linux-android-4.9` (in branch  `android10-release`) ,  those error messages has been printed by make command. How can I deal this error?
```
../scripts/Makefile.build:671: recipe for target 'drivers/misc/mediatek' failed
make[3]: *** [drivers/misc/mediatek] Error 2
../scripts/Makefile.build:671: recipe for target 'drivers/misc' failed
make[2]: *** [drivers/misc] Error 2
/home/ubuntu/project/testAnd/Xiaomi_Kernel_OpenSource/Makefile:1115: recipe for target 'drivers' failed
make[1]: *** [drivers] Error 2
```

OS version:  `VERSION="18.04.4 LTS (Bionic Beaver)"`
gcc version: `gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0`

rama982, 2021-12-30:
Use clang compiler

rama982, 2021-12-30:
Also use one defconfig