MiCode/Xiaomi_Kernel_OpenSource #2371: Error while compiling dandelion-q-oss kernel

thofyb, 2022-01-31:
I downloaded the branch with the kernel and linaro toolchain
Wrote the following commands in the terminal:
```
mkdir out
export ARCH=arm64
export SUBARCH=arm64
export CROSS_COMPILE=${PWD}/toolchain/bin/aarch64-linux-gnu-
make O=out dandelion_defconfig
make -j$(nproc) O=out 2>&1 | tee kernel.log
```
After the last make command, I got the following error: 
```
make[1]: Entering directory '/home/thofyb/Xiaomi_Kernel_OpenSource/out'
  CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
for i in ../arch/arm64/boot/dts/mediatek/dandelion.dts; do \
        base_prj=`grep -m 1 '#include [<\"].*\/cust\.dtsi[>\"]' $i | sed 's/#include [<"]//g'\
        | sed 's/\/cust\.dtsi[>"]//g' | sed 's/\/\*//g' | sed 's/\*\///g' | sed 's/ //g'`\
        prj_path=./arch/arm64/boot/dts/$base_prj ;\
        dws_path=../drivers/misc/mediatek/dws/mt6765/$base_prj.dws ;\
        if [ -f $dws_path ] ; then \
                mkdir -p $prj_path ;\
                 ../tools/dct/DrvGen.py $dws_path $prj_path $prj_path cust_dtsi;\
        fi \
done
for i in ../arch/arm64/boot/dts/mediatek/dandelion.dts; do \
        base_prj=`grep -m 1 '#include [<\"].*\/cust\.dtsi[>\"]' $i | sed 's/#include [<"]//g'\
        | sed 's/\/cust\.dtsi[>"]//g' | sed 's/\/\*//g' | sed 's/\*\///g' | sed 's/ //g'`\
        prj_path=./arch/arm64/boot/dts/$base_prj ;\
        dws_path=../drivers/misc/mediatek/dws/mt6765/$base_prj.dws ;\
        if [ -f $dws_path ] ; then \
                mkdir -p $prj_path ;\
                 ../tools/dct/DrvGen.py $dws_path $prj_path $prj_path cust_dtsi;\
        fi \
done
/bin/sh: 8: ../tools/dct/DrvGen.py: not found
make[1]: *** [../scripts/drvgen/drvgen.mk:63: arch/arm64/boot/dts/dandelion/cust.dtsi] Error 127
make[1]: Leaving directory '/home/thofyb/Xiaomi_Kernel_OpenSource/out'
make: *** [Makefile:152: sub-make] Error 2
```
What is the error and how to fix it?

hendramanudinata03, 2022-02-02:
Try to run the Python script directly:

```bash
./tools/dct/DrvGen.py
```

What's the output?

Usually this error pops out because missing required package(s) to execute, like, in this case, Python. Try to install it and see if it works or not.

I've compiled the Kernel recently, and so far no error like that. Here's my source: https://github.com/hendramanudinata03/HenHunter_xiaomi_dandelion/tree/HenHunter-BETA.

thofyb, 2022-02-03:
@hendramanudinata03 Thank you for the comment! It turned out that I didn't install python 2 as python.
I'm glad you wrote this comment, because I desperately thought that no one would respond and I would fight all the problems on my own. So if there are any big problems, then I will consult with you, ok?

closed 2022-02-03 by thofyb

hendramanudinata03, 2022-02-03:
Yeah sure!