MiCode/Xiaomi_Kernel_OpenSource #2381: drivers/misc/Kconfig:540: can't open file "drivers/misc/hwid/Kconfig"

PShocker, 2022-02-22:
按照Wiki里的教程编的,但是并没有这个文件,我看源码里也没有,啥情况?

PShocker, 2022-02-23:
../drivers/thermal/thermal_core.c:27:10: fatal error: trace/hooks/thermal.h: No such file or directory
 #include <trace/hooks/thermal.h>
还有include/trace/hooks/thermal.h也没有

xiufeiwu, 2023-03-28:
他们源码里把drivers/misc/hwid/给去掉了，我在github全站搜了下，找到一个项目https://github.com/Xiaomi-Diting-Development/android_kernel_xiaomi_diting 这个项目里复制一下drivers/misc/hwid/到你的项目里，然后就可以了

ccccoisini, 2023-06-30:
> 按照Wiki里的教程编的，但是并没有这个文件，我看源码里也没有，啥情况？

你好,请问你后来解决了吗

kuailexs, 2023-09-07:
> ../drivers/thermal/thermal_core.c:27:10: fatal error: trace/hooks/thermal.h: No such file or directory #include <trace/hooks/thermal.h> 还有include/trace/hooks/thermal.h也没有

 在谷歌的原版内核里复制这个文件就可以了

YunMoYYDS, 2026-08-28:
> 他们源码里把drivers/misc/hwid/给去掉了，我在github全站搜了下，找到一个项目https://github.com/Xiaomi-Diting-Development/android_kernel_xiaomi_diting 这个项目里复制一下drivers/misc/hwid/到你的项目里，然后就可以了

这个仓库已经404了，我看别人说谷歌的通用内核拆下来能装上去，是真的吗？