MiCode/Xiaomi_Kernel_OpenSource #2383: 【request】请更新编译说明及环境配置安装

zixijian, 2022-02-27:
按照https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone做完全不行
1）gcc4.9都被谷歌移除了，谷歌clang才6.0，clang15都出了
2）自行使用clang能出内核不开机，不开机连个原因都不知道
3）violet是4.14内核，设备树没分开吧，为什么编译不出来或不开机
4）干点人事吧

zixijian, 2022-02-27:
> 大哥我也被小米这个wiki坑惨了！！！你怎么用clang编译的，指点一下行吗？

我写了一篇在手机上编译的文章，现在出来的内核都不开机了，也不知道小米更新了什么。
https://zixijian.github.io/2021/01/15/008.html

ABCCCC55645687, 2022-02-27:
好的感谢大哥，小米这破玩意这开了个**的源，不愿意开就别开，草了，这内核源码是个这就算了，隔壁的patchrom都烂尾了

Angel0726, 2022-03-03:
> 好的感谢大哥，小米这破玩意这开了个**的源，不愿意开就别开，草了，这内核源码是个这就算了，隔壁的patchrom都烂尾了

希望小米在开源代码以及分支中添加how_to_compile文件。以此指导每一个内核的编译环境与编译工具。类似于这种
[image]

ABCCCC55645687, 2022-03-04:
> > 好的感谢大哥，小米这破玩意这开了个**的源，不愿意开就别开，草了，这内核源码是个这就算了，隔壁的patchrom都烂尾了
> 
> 希望小米在开源代码以及分支中添加how_to_compile文件。以此指导每一个内核的编译环境与编译工具。类似于这种 [image]
“别人家的内核源码:怕你不会跑，特地的整了许多的编译教程和查错指南”
“啸米的源码:就怕你能编译出一个东西”

ABCCCC55645687, 2022-03-06:
> > 大哥我也被小米这个wiki坑惨了！！！你怎么用clang编译的，指点一下行吗？
> 
> 我写了一篇在手机上编译的文章，现在出来的内核都不开机了，也不知道小米更新了什么。 https://zixijian.github.io/2021/01/15/008.html

大哥！太感谢您的Clang编译教程了！我用了您的教程成功出文件！能开机，不好之处是WLAN和声音不知为什么坏了。。。不过我知足了，毕竟我编译内核只是为了玩玩KVM而已，哈哈。
再次感谢您！

Angel0726, 2022-03-08:
> > > 大哥我也被小米这个wiki坑惨了！！！你怎么用clang编译的，指点一下行吗？
> > 
> > 
> > 我写了一篇在手机上编译的文章，现在出来的内核都不开机了，也不知道小米更新了什么。 https://zixijian.github.io/2021/01/15/008.html
> 
> 大哥！太感谢您的Clang编译教程了！我用了您的教程成功出文件！能开机，不好之处是WLAN和声音不知为什么坏了。。。不过我知足了，毕竟我编译内核只是为了玩玩KVM而已，哈哈。 再次感谢您！

我也是为了kvm。但是不能用wlan怎么联网。你用的哪款手机？高通还是联发科

zixijian, 2022-03-09:
> > > > 大哥我也被小米这个wiki坑惨了！！！你怎么用clang编译的，指点一下行吗？
> > > 
> > > 
> > > 我写了一篇在手机上编译的文章，现在出来的内核都不开机了，也不知道小米更新了什么。 https://zixijian.github.io/2021/01/15/008.html
> > 
> > 
> > 大哥！太感谢您的Clang编译教程了！我用了您的教程成功出文件！能开机，不好之处是WLAN和声音不知为什么坏了。。。不过我知足了，毕竟我编译内核只是为了玩玩KVM而已，哈哈。 再次感谢您！
> 
> 我也是为了kvm。但是不能用wlan怎么联网。你用的哪款手机？高通还是联发科

内核4.19以上已经分离驱动dtbo.img，和内核不在一起。可以看一下其他现成的内核，不要用官网的。

ABCCCC55645687, 2022-03-09:
> > > > 大哥我也被小米这个wiki坑惨了！！！你怎么用clang编译的，指点一下行吗？
> > > 
> > > 
> > > 我写了一篇在手机上编译的文章，现在出来的内核都不开机了，也不知道小米更新了什么。 https://zixijian.github.io/2021/01/15/008.html
> > 
> > 
> > 大哥！太感谢您的Clang编译教程了！我用了您的教程成功出文件！能开机，不好之处是WLAN和声音不知为什么坏了。。。不过我知足了，毕竟我编译内核只是为了玩玩KVM而已，哈哈。 再次感谢您！
> 
> 我也是为了kvm。但是不能用wlan怎么联网。你用的哪款手机？高通还是联发科

源码分支:laurus-q-oss
设备全名:MI CC 9e
处理器:高通 Snapdragon 665

ABCCCC55645687, 2022-03-09:
> > > > 大哥我也被小米这个wiki坑惨了！！！你怎么用clang编译的，指点一下行吗？
> > > 
> > > 
> > > 我写了一篇在手机上编译的文章，现在出来的内核都不开机了，也不知道小米更新了什么。 https://zixijian.github.io/2021/01/15/008.html
> > 
> > 
> > 大哥！太感谢您的Clang编译教程了！我用了您的教程成功出文件！能开机，不好之处是WLAN和声音不知为什么坏了。。。不过我知足了，毕竟我编译内核只是为了玩玩KVM而已，哈哈。 再次感谢您！
> 
> 我也是为了kvm。但是不能用wlan怎么联网。你用的哪款手机？高通还是联发科

忘了说了，流量可以驱动的，而且我这卡流量无限量，哈哈。
不过Wiki倒是给出了编译WLAN驱动的教程，又需要用到那个恶心的代码激光论坛(CodeAurora)

lateautumn233, 2022-03-09:
请不要再尝试无意义的事情，在高通平台kvm被禁用了的
至于wifi驱动什么的自己用脚本导入吧

ABCCCC55645687, 2022-03-09:
> 请不要再尝试无意义的事情，在高通平台kvm被禁用了的 至于wifi驱动什么的自己用脚本导入吧

啊？这...行吧，那算了

Angel0726, 2022-03-10:
> 请不要再尝试无意义的事情，在高通平台kvm被禁用了的 至于wifi驱动什么的自己用脚本导入吧
我有三个问题：
1. 联发科的kvm可以使用吗？
2. 是小米的高通平台禁用了，还是所有手机都禁用了？realme的高通kvm可以使用吗？
3. 高通禁用了kvm为什么跟wifi驱动有关。kvm应该是不能开内核虚拟机吧