MiCode/Xiaomi_Kernel_OpenSource #2390: [Request] kernel source for Redmi Note 11

shashankx86, 2022-03-12:


spi3ex, 2022-03-19:
Following this Post as I too Own Note 11 and waiting for Source code,

warism100, 2022-03-21:
so without source code no custom roms ? no lineageos  , pixel experience ,right ?

warism100, 2022-03-25:
hi the information you write here  https://github.com/shashankx86/android_device_xiaomi_spes   
about the processor is note correct its an snapdragon 680 net 660 right ?

Schmuppes, 2022-07-07:
Releasing the kernel source is needed, we're waiting.

jkriyabd, 2022-07-31:
after releasing note 11 has passed 8 month around but still we are not getting the source code, 
why xiaomi?  its shame on you!

jkriyabd, 2022-08-06:
any update for releasing

shashankx86, 2022-08-06:
> any update for releasing

Nope

Vagelis1608, 2022-08-18:
Isn't it a requirement of AOSP's license to release the sources?
Or so I thought, anyway

Ilya0Walker, 2022-08-31:
waiting for source code.

Riqqq01, 2022-10-30:
Please release the kernel source for redmi note 11....please!!!

Iaonnis, 2022-11-24:
Xioami please release this thing fast, almost a year now.

mpxbroadcast, 2022-12-31:
Where is the damm kernel xiaomi. we want to build it

Vagelis1608, 2023-01-01:
It would be a lot cooler if you stopped violating the AOSP license

shashankx86, 2023-01-01:
> It would be a lot cooler if you stopped violating the AOSP license

let's just sue them

azka033, 2023-03-15:
when will the redmi note 11 kernel be released please quickly remove it like other devices?

azka033, 2023-03-16:
Menunggu sumber