MiCode/Xiaomi_Kernel_OpenSource #2419: [Request] Kernel source for Global Xiaomi 11T (Agate) MTK

BayuMustikaIP, 2022-04-17:
Kernel Source Not Release Since 6 month this device launch, please release it

kidukarai, 2022-04-28:
Please Xiaomi Mi 11T (agate) is an amazing device with a very impressive MTK processor please release the Mi11T (agate) source code :) :+1:

ImRyujinnn, 2022-06-06:
@MiCode

SAddik13, 2022-06-26:
they are not responding

cata332, 2022-07-09:
@MiCode

ImRyujinnn, 2022-07-09:
> @MiCode @ImRyujinnn Are you manage to build https://github.com/ImRyujinnn/recovery_device_xiaomi_agate ? **Please relase the recovery !**

it cannot booting into system, but it booted to twrp

cata332, 2022-07-09:
@ImRyujinnn Please let me test your recovery...maibe with a custom rom?
Thank you

ImRyujinnn, 2022-07-09:
> @ImRyujinnn Please let me test your recovery...maibe with a custom rom? Thank you

https://github.com/ImRyujinnn/Action-Recovery-Builder/releases/download/2620231118/boot.img , if you reboot into system it will booted into twrp again

cata332, 2022-07-09:
Android 11 or 12?

ImRyujinnn, 2022-07-09:
@cata332 android 12 miui 13.0.1.0 ID

bachtiyar93, 2022-11-11:
Why not have response after thread 11 months created?

caezaar, 2022-12-21:
please xiaomi respond !!

Stephanizgo, 2023-03-09:
It's well over 15 months still no kernel

mansoor4455, 2023-03-28:
The biggest mistake in my life is that I chose the Xiaomi device.

jyuuren, 2023-04-29:
So many people using this phone and still no kernel. Dear Xiaomi PLEASE release kernel source for this device!

bachtiyar93, 2023-05-16:
If no have commited by @mi-code  this is last my phone using xiaomi brand.

popy2k14, 2023-05-16:
Mi too 😬

EvoTeamMalaysia, 2023-05-22:
> If no have commited by @mi-code this is last my phone using xiaomi brand.

Yeah ,the reason we usually buy xiaomi phone is because of easy access of twrp and rom flashing and other kind of things to play with it ,but since u arent giving the kernel source ,how the heck we are able to do it

Stephanizgo, 2023-07-29:
Mediatek is brand friendly but any consumer meaning if you buy this phone with mediatek soc, you don't own it because Xiaomi tried to save money on sources. This phone is not very GSI friendly device, GSIs don't boot sometimes and most of the times power button doesn't work on GSIs. We need roms to make this device work and feel fresh again. I'm bored with MIUI, especially the global version. Xiaomi.eu is okay but this device didn't sell in china, which means that we can't get xiaomi.eu on this device (at least not for free), or other Xiaomi.eu based roms like MIUI mind edition ect. I'll hopefully get a new device this year, but I still got hope that they'll release the sources once last OTA arrives or something on this device (could take 2 more years) but I don't recommend getting Mediatek devices at the moment