MiCode/Xiaomi_Kernel_OpenSource #2468: [Request] Kernel Source Code for Redmi Note 11 Pro 5G (veux)

roema, 2022-05-13:


osebee, 2022-06-12:
+1
where is kernel source for redmi note 11 pro 5G global?

SaddikHussain786, 2022-06-20:
Please release Redmi Note 11 Pro 5G (Qualcomm SM6375) Kernel source

roema, 2022-07-28:
Please comply by GPL License of kernel.

roema, 2022-07-28:
If everyone is intrested, ArrowOS available for veux

https://arrowos.net/download

Poco Veux

Xiaomi roms are just a disaster...

mdtameemkhan, 2022-10-05:
+1
Bump

mikefragg, 2023-01-01:
It's GPL Violation to not release the Kernel. I would highly appreciate if the Kernel Code is released as soon as possible.

PJB1906, 2023-01-15:
> Please release Redmi Note 11 Pro 5G (Qualcomm SM6375) Kernel source

> Please release Redmi Note 11 Pro 5G (Qualcomm SM6375) Kernel source

Kindly release the kernel source code for veux

saharenil, 2023-02-01:
xiaomi not following rules by not giving kernel source code for VEUX INDIAN Version....
pls give us kernel source code

PJB1906, 2023-02-13:
> xiaomi not following rules by not giving kernel source code for VEUX INDIAN Version.... pls give us kernel source code as soon as possible

Can u tell me which Rule or Agreement ??

yoshimo, 2023-02-13:
GPL , the code license

mdtameemkhan, 2023-05-05:
someone should sue them already

satyam9416, 2023-05-14:
> Please release Redmi Note 11 Pro 5G (Qualcomm SM6375) Kernel source

+1

djprmf, 2023-06-12:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/veux-r-oss