MiCode/Xiaomi_Kernel_OpenSource #2491: [BUMP] Redmi Note 9 5G (cannon) kernel source is missing

LinkBoi00, 2022-06-10:
@mi-code 
Are you guys planning on releasing it or what? It's been ~1 month now since MIUI 13 dropped.

LinkBoi00, 2022-06-10:
@ivan19871002 Any input?

Ichigokano, 2022-06-17:
All new device's source haven't been release.Even qcom devices.May be they are busy for building cars

ivan19871002, 2022-06-19:
> @ivan19871002 Any input?

Sorry, I left xiaomi.

MrArtemSid, 2022-06-19:
@ivan19871002 sad, btw why Xiaomi didn't push Redmi 4 (prada) source?

ShreyDotExe, 2022-06-23:
> 

So, i guess we never getting any kernel sources for veux aswell?

ivan19871002, 2022-06-24:
> 
sorry， I do not know this .. I left xiaomi.

ajaikumarnadar, 2022-07-15:
Hi @ivan19871002 ,
Anyone person we can reach out ?about this kernel source, many devices are missing kernel sources from Xiaomi.
Hope you can help...

Thanks

ivan19871002, 2022-07-15:
> 

Since I left, I haven't found someone responsible

ajaikumarnadar, 2022-07-15:
ok, Thanks for the reply @ivan19871002.
It looks like Xiaomi is feeling a hard time finding your replacement 😂😂😂

ivan19871002, 2022-07-20:
> 

lol

mikefragg, 2023-01-01:
It's GPL Violation to not release the Kernel. I would highly appreciate if the Kernel Code is released as soon as possible.