MiCode/Xiaomi_Kernel_OpenSource #2497: Violating GNU GENERAL PUBLIC LICENSE !!!

willzyx-hub, 2022-06-18:
Hello Xiaomi i just see that you have not updated camellia-s-oss / spes-r-oss for a long time please note that you are using GPL License which mean you must release the source code ! You are now violating GPL License now please release it ASAP !

xslvrxslwt, 2022-06-19:
The what hahahahah, they're violating nothing you stupid Indian.

willzyx-hub, 2022-06-19:
> The what hahahahah, they're violating nothing you stupid Indian.

Im not Indian and if you think this them sorry :)

Harshan-Dinuka-Gunarathne, 2022-07-10:
> The what hahahahah, they're violating nothing you stupid Indian.

racist idiot .

Divyanshu-Modi, 2022-07-10:
> The what hahahahah, they're violating nothing you stupid Indian.

If they are using the opensource code and not publicly releasing it than it is a GPL violation even a zip is fine but they are straight up dead

Divyanshu-Modi, 2022-07-10:
> > The what hahahahah, they're violating nothing you stupid Indian.
> 
> Im not Indian and if you think this then sorry :) Yes i Recheked it Xiaomi use GNU General Public License which have no right to keep the source code hidden :)

The maintainer is out of ximi so unless a new maintainer is appointed this issues are useless