MiCode/Xiaomi_Kernel_OpenSource #2513: Evo x unofficial frost rom on poco m3

Krishnadude678, 2022-06-29:
The mobile network is not working on Google apps like playstore YouTube chrome etc but wifi is working I tried restart airplane mod everything but not solved and an exclamation mark is also coming in right side of my network
[image]

zainarbani, 2022-06-29:
just what the fuck

Krishnadude678, 2022-06-30:
Bro help me if u know the solution

RayapKun, 2022-07-01:
LOLLLLLLLLL

RayapKun, 2022-07-01:
funny to reporting rom bug at micode kernel died source

Subhendu-droid, 2022-07-04:
LMAO WHAT

ibratabian17, 2022-07-04:
LMFAO bruh lol

imannig, 2022-07-04:
@frostg012

imannig, 2022-07-04:
@frostg-012 help sar

Krishnadude678, 2022-07-04:
> funny to reporting from bug at mico de kernel died source

Hehe I am new at GitHub ... A guy gives me the advice to write bugs at GitHub 😂

Sushrut1101, 2022-07-04:
Lmfao

Krishnadude678, 2022-07-04:
> 

Are you a pro developer

is-nrdra, 2022-07-04:
@frostg-012 plz help sir

DrtSinX98, 2022-07-04:
*Dies of cringe*

Subhendu-droid, 2022-07-04:
> > 
> 
> Are you a pro developer

Yes he's very Peru ultra pro Max debheloper :)

NeonzGamingYT, 2022-07-04:
lmao what? 🤣

sayann70, 2022-07-04:
lmaooo deddddd ☠️

iamthecloverly, 2022-07-04:
Bruh lamo🤣🤣

keneankit01, 2022-07-04:
Bruuuuu 🤣🤣🤣🤣 Epic place to report

TwistedPrime, 2022-07-04:
> > funny to reporting from bug at mico de kernel died source
> 
> Hehe I am new at GitHub ... A guy gives me the advice to write bugs at GitHub 😂 btw which language u prefer for making rom

HTML

Krishnadude678, 2022-07-04:
> > > funny to reporting from bug at mico de kernel died source
> > 
> > 
> > Hehe I am new at GitHub ... A guy gives me the advice to write bugs at GitHub 😂 btw which language u prefer for making rom
> 
> HTML

Stop kidding with me bro😂🤣

Krishnadude678, 2022-07-04:
> Bruuuuu 🤣🤣🤣🤣 Epic place to report
Bruh this is serious GitHub repo not a Kapil Sharma show to laugh maintain decency😂😂🤣

Krishnadude678, 2022-07-04:
> _Dies of cringe_

😂😂

Krishnadude678, 2022-07-04:
> Lmao 🤣 sir are u serious?🤣🤣🤣

What do u think 🤣😂

Playez, 2022-07-06:
helo im ximi indian developer. we at oyr copmany develop a profesional fix to ur tissue,. donwload nusatrnaata os adn it would it will fix your problemo. 
goobyed

Surax98, 2022-07-09:
Gr8 b8 m8, I r8 it 8/8