MiCode/Xiaomi_Kernel_OpenSource #2525: Please release kernel source for Redmi Note 11T pro (xagapro,xaga)

lyhooo, 2022-07-09:


ajaikumarnadar, 2022-07-14:
It would be good if you guys complained about this GPLv2 violation here. 
https://www.gnu.org/licenses/gpl-violation.en.html

dyt1993, 2022-07-27:
Please release kernel source for Redmi Note 11T pro    THANK YOU!

NextWork123, 2022-09-01:
Come on boys relase this kernel source :D

Shubhakashyap, 2022-09-03:
Come on boys release kernel source for redmi k50i humble request  to xiaomi

AnyOne06, 2022-09-13:
Xiaomi so slow realme released their kernel source in Dimensity8100 phones already 2 months ago 😅

D0CT0RFATE, 2022-10-11:
@Xiaomi Kindly release kernel sources for Redmi K50i (XAGA) with Dimensity 8100 processor. It's been over 2 full months since the release of this device and it's high time you guys allow developers to work in the device and make use of it's full potential! Pls release the kernel sources ASAP!
Thank you.

Deathwish98, 2023-05-17:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/xaga-s-oss

Isn't this the link for kernel source ?