MiCode/Xiaomi_Kernel_OpenSource #2536: Xiaomi violates the GNU license

bimizx, 2022-07-24:
It's understandable that Xiaomi is making money, but time and time again it doesn't release the kernel source code. Copy someone else's work, make a profit, and then give nothing in return. It's been seven months without an update since they last updated the device.It's a monopoly.

The Linux kernel used by Android is GPL-compliant, which means Xiaomi must publish the source code.And the source code released by Xiaomi is incomplete, he has no right to hide the source code.And he only releases the source code once for a mobile phone, disgusting Xiaomi.

Although Xiaomi's infringement does not seem to have much impact on the average Android user, at the development level, it has caused serious damage to the entire ecosystem, as the GNU project illustrates, in this case, the hinder the development of the entire software industry.

Finally, require Xiaomi to abide by the GPL agreement to open source the kernel on time.

bimizx, 2022-07-24:
Fuck you Xiaomi

Molyuu, 2022-07-24:
Yes!I stand with you dude

Molyuu, 2022-07-24:
A lot of issues are started with "please release xxxxx".But guys,remember,Xiaomi SHOULD AND MUST ALLOW THE GNU LICENSE AND OPEN SOURCE THEIR KERNEL.

sadoyui, 2022-07-25:
Fuck you Xiaomi

Molyuu, 2022-07-25:
So Xiaomi,fuck you!

ThieuMinh26, 2022-07-27:
will device shipped with gki fix this issue ?

ashley0143, 2022-07-29:
hello, ashley from free software foundation here :3

we are going to talk about this to Xiaomi, we still dont know why they dont publish the full source code.

k bye

iGOR121, 2022-07-29:
> hello, ashley from free software foundation here :3
> 
> we are going to talk about this to Xiaomi, we still dont know why they dont publish the full source code.
> 
> k bye

isn't it obvious that the guy who worked on it just left ximi? no-one's left to do it... just wait for them to get another one

Molyuu, 2022-07-30:
No.According to the lefted guy who was considered as the head on Weibo(A
Chinese social media platform),he did nothing.

GEOEGII555, 2025-02-07:
I have contacted Xiaomi customer support, they said "we are bonded to the terms and conditions of the company policies". I'm assuming that they meant MediaTek's policy. Tho if it is MediaTek stopping them from publishing the code - does that mean MediaTek is (pretty sure they're doing it knowingly) breaking the GPL?