MiCode/Xiaomi_Kernel_OpenSource #2565: [Request] For god sake, Release the kernel source Redmi Note 11 (spes/spesn)

jkriyabd, 2022-08-01:
plz sir,
we are waiting for long time, without source we are unable to make stable custom rom, please release the source we are all rn11 lover waiting for long plz plz plz 😥

jbg, 2022-08-03:
You don't have to say please, they have a legal obligation to release it.

jkriyabd, 2022-08-03:
yeah right but what should we do???,, without custom kernel device is not handy and limited

jbg, 2022-08-03:
You can wait for them to do it willingly, or you can find a copyright holder (anyone who has contributed code to the Linux kernel) and convince/fund them to take Xiaomi to court to enforce the GPL. sfconservancy.org or gpl-violations.org may be able to help with the latter approach.

willzyx-hub, 2022-08-07:
NOT GONNA HAPPEND

jkriyabd, 2022-08-07:
> NOT GONNA HAPPEND

Can you explain bro?

Dedicatus5457, 2022-08-09:
> You can wait for them to do it willingly, or you can find a copyright holder (anyone who has contributed code to the Linux kernel) and convince/fund them to take Xiaomi to court to enforce the GPL. sfconservancy.org or gpl-violations.org may be able to help with the latter approach.

Bro, you're thinking too easily. In China, the GPL has basically no enforcement effect, and Chinese judges do not know what the GPL is.

jkriyabd, 2022-08-09:
so in this circumstances what should we do bro?

jbg, 2022-08-10:
> > You can wait for them to do it willingly, or you can find a copyright holder (anyone who has contributed code to the Linux kernel) and convince/fund them to take Xiaomi to court to enforce the GPL. sfconservancy.org or gpl-violations.org may be able to help with the latter approach.
> 
> 
> 
> Bro, you're thinking too easily. In China, the GPL has basically no enforcement effect, and Chinese judges do not know what the GPL is.There is also no organization like the Free Software Foundation that can provide u with legal aid

China has nothing to do with it. You can bring a case in any jurisdiction where Xiaomi sells an infringing device. Pick one with a modern legal system and decent Xiaomi sales so that they're motivated not to just pull out.

willzyx-hub, 2022-08-10:
> China has nothing to do with it. You can bring a case in any jurisdiction where Xiaomi sells an infringing device. Pick one with a modern legal system and decent Xiaomi sales so that they're [...]

Yes, Xiaomi could be reported for this.