MiCode/Xiaomi_Kernel_OpenSource #2572: Kernel source - Veux

JaswalAshish48, 2022-08-10:
Release kernel source for Veux please

SweatMonster, 2022-08-10:
At least tell us about why hasn't it released yet and the future plans regarding the kernel sources.

dolaijagadish, 2022-08-10:
Release source code please.... @mi-code

god-goku, 2022-08-10:
Give source

SauRavRwT, 2022-08-10:
Release kernel source for Veux

PenguinSlayerzz, 2022-08-10:
source plz

tamimmahmudkhan, 2022-08-10:
Please release kernel sources for your peux/veux line of devices. Devices released after have their sources already. Please release it for our devices. Or at least give an eta on when sources will surely be released.

SharmagRit, 2022-08-10:
Xiaomi, when releasing kernel sources for veux, it's has been more than 6months after release, without kernel sources future development is stuck...

charlieangriawan, 2022-08-10:
yes, that would be great

dolaijagadish, 2022-08-11:
Please do something for us

george-andrei, 2022-08-11:
Can anyone from Xiaomi comment on this?

mikefragg, 2023-01-01:
It's GPL Violation to not release the Kernel. I would highly appreciate if the Kernel Code is released as soon as possible.

amritokarmokar69, 2023-01-27:
Release Kernel sources of veux

AdrianRR, 2023-01-27:
Release sources Kernel veux

krushnadeore, 2023-01-29:
kernel source for Veux ( poco x4 pro indian) wen release ?

samduploader, 2023-05-14:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/4147

AmeyaGurjar, 2023-05-16:
Please Release The Kernel Source For Veux..