MiCode/Xiaomi_Kernel_OpenSource #2575: munch-s-oss Missing techpack code for display,camera and audio

Instinct369, 2022-08-10:


Aquarius223, 2022-08-11:
audio is https://github.com/MiCode/vendor_qcom_opensource_audio-kernel/tree/munch-s-oss

VaibhavTakVT, 2022-08-11:
Codes for Display and camera is not there

akash07k, 2022-08-11:
@Aquarius223  Any idea that where is the code for Camera and Display?
Also, @mi-code Why the Munch branch was deleted/pulled back? If there was any issue, then obviously it was possible to fix it with a commit.

OptimBro, 2022-08-12:
@mi-code Why the Munch branch was deleted/pulled back? If there was any issue, then obviously it was possible to fix it with a commit.

Cleveland996, 2022-08-19:
> @mi-code为什么 Munch 分支被删除/撤回？如果有任何问题，那么显然可以通过提交来修复它。

He may have open-sourced something they shouldn't have open-sourced...

akash07k, 2022-08-19:
@Cleveland996  And many people have the entire dump of that with them including me!

PaiJi, 2022-08-28:
them publish it again!
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/munch-s-oss

akash07k, 2022-08-28:
Wonderful.

Are there any changes too?

PaiJi, 2022-08-28:
@akash07k Them delete all files from folder name "mihw" which location drivers.