MiCode/Xiaomi_Kernel_OpenSource #2583: Request open source kernel for POCO F4&K40s(munch)

Tingting-coder, 2022-08-14:
I bought POCO phone for flash other system. But I find that Xiaomi doesn't support for kernel open source these days. So I request for open source for munch on Github. If the team can see that, please open source! Many users hope that !

Cleveland996, 2022-08-14:
Agree with you！！

VaibhavTakVT, 2022-08-14:
@mi-code please release source code for poco f4

OptimBro, 2022-08-16:
@mi-code please release source code for poco f4

psklf, 2022-08-18:
please release source code for poco f4

Aquarius223, 2022-08-19:
Redmi K40S Android S kernel Source:
* Note, the techpack lacks display and camera.
kernel: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/3d665367f026a
audio-kernel: https://github.com/MiCode/vendor_qcom_opensource_audio-kernel/tree/munch-s-oss

OptimBro, 2022-08-20:
> Redmi K40S Android S kernel Source:
> 
> * Note, the techpack lacks display and camera.
>   kernel: [3d665367f026a](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/3d665367f026a)
>   audio-kernel: https://github.com/MiCode/vendor_qcom_opensource_audio-kernel/tree/munch-s-oss

Can we use the source of alioth?

PaiJi, 2022-08-28:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/munch-s-oss

closed 2022-12-11 by Tingting-coder