MiCode/Xiaomi_Kernel_OpenSource #2585: Help wanted!

Saikatsaha1996, 2022-08-15:
Any solution for this ?

AR      built-in.a
  MODPOST vmlinux.o
WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
ld: drivers/platform/msm/gsi/gsi.o: relocation R_AARCH64_ABS32 against `__crc_gsi_write_channel_scratch' can not be used when making a shared object
drivers/platform/msm/gsi/gsi.o:(.data+0x0): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(.data+0x28): dangerous relocation: unsupported relocation
drivers/platform/msm/gsi/gsi.o:(.data+0x50): dangerous relocation: unsupported relocation
make[1]: *** [/home/userland/android_kernel_xiaomi_sm8250/Makefile:1154: vmlinux] Error 1
make[1]: Leaving directory '/home/userland/ZenMaxBuilder/out/apollo'
make: *** [Makefile:146: sub-make] Error 2

Sentrymon, 2022-08-17:
I'm not well acquainted with the code base but that looks like some elevation problem where the user don't posses root privileges.

Saikatsaha1996, 2022-08-17:
> I'm not well acquainted with the code base but that looks like some elevation problem where the user don't posses root privileges.

Solve with patch

closed 2022-08-17 by Saikatsaha1996