MiCode/Xiaomi_Kernel_OpenSource #2602: fw_sample.i  缺fw_sample.i 文件，编译失败

yybbll, 2022-08-31:
merlin-r-oss  缺fw_sample.i 文件，编译失败

erikeah, 2023-01-28:
Same here! Upload the drivers blobs.

jiawe, 2023-02-01:
copy this file  https://github.com/AOSPA/android_kernel_xiaomi_laurel_sprout/blob/373fd3abc3203d8201b696a3e8dceab30897ba84/drivers/input/touchscreen/focaltech_touch/include/firmware/fw_sample.i  to the drivers/input/touchscreen/focaltech_touch/include/firmware/

jiawe, 2023-02-01:
> Same here! Upload the drivers blobs.

copy this file https://github.com/AOSPA/android_kernel_xiaomi_laurel_sprout/blob/373fd3abc3203d8201b696a3e8dceab30897ba84/drivers/input/touchscreen/focaltech_touch/include/firmware/fw_sample.i to the drivers/input/touchscreen/focaltech_touch/include/firmware/