MiCode/Xiaomi_Kernel_OpenSource #2616: Device Xaga/Redminote11tpro/X4GT/K50i /Request/Open source/ForMtkD8100

AnyOne06, 2022-09-10:
As for this device development and more and more users that wishes to get an open source device I'm begging for your support and I'm hopefully requesting this device kernel source for better customization.Plss don't leave mtk behind of beauty and passion on custom roms.
I'm gay don't be racist!

AnyOne06, 2022-10-13:
Ty xioami

AnyOne06, 2022-10-13:
Ty

closed 2022-10-13 by AnyOne06

ParampreetR, 2022-10-25:
@AnyOne06 you seem professional, they released kernel source for `xaga`. So will it be enough to develop custom roms for Redmi k50i/note 11T pro? Coz I have heard that we can't expect to see custom roms for devices with mediatek chipset. In short, can we expect to see custom roms for these devices? Noob here

AnyOne06, 2022-10-25:
> @AnyOne06 you seem professional, they released kernel source for `xaga`. So will it be enough to develop custom roms for Redmi k50i/note 11T pro? Coz I have heard that we can't expect to see custom [...]

Yes very soon ☺️ something is coming

ParampreetR, 2022-10-25:
> Yes very soon ☺️ something is coming

I'll be waiting 😃 and thanks for quick reply

yaar007, 2022-12-11:
> Yes very soon ☺️ something is coming

Are you working on xaga/xagain soon?

Samandar02, 2024-04-26:
@ParampreetR
What I heard is it is possible to build a custom ROM, but the problem is on availability of community (supportive) for MTK devices in case something goes wrong, but there are MIUI EU 13 and 14 ROMs available to download.
In my experience, MTK has always been the most  to be unbrickable