MiCode/Xiaomi_Kernel_OpenSource #2617: [REQUEST] Release xiaomi 12 (cupid) bootloader

eclipsek20, 2022-09-11:
As stated in the title releasing the bootloader would be very nice

HyperN00B, 2022-09-11:
And you really think Xiaomi will just release the code of the bootloader like that?

eclipsek20, 2022-09-11:
> And you really think Xiaomi will just release the code of the bootloader like that?

Well it's better to have hope than to not have hope.

xenxynon, 2022-09-11:
He has a point

Sushrut1101, 2022-09-11:
> > And you really think Xiaomi will just release the code of the bootloader like that?
> 
> Well it's better to have hope than to not have hope.

Lmao
Spam Xiaomi to release the vendor/qcom too 😂

dakkshesh07, 2022-09-11:
> > > And you really think Xiaomi will just release the code of the bootloader like that?
> > 
> > 
> > Well it's better to have hope than to not have hope.
> 
> Lmao Spam Xiaomi to release the entire vendor/qcom 😂

if going till that extent, why left mtk behind... Release that also.

NotEclypse, 2022-09-11:
> He has a point

yeah hes got a point there

HyperN00B, 2022-09-11:
Why not just release the code for every blob in every Xiaomi device and a bootloader code that doesn't verify the boot process and lets you load custom stuff with locked BL?

HyperN00B, 2022-09-11:
> > And you really think Xiaomi will just release the code of the bootloader like that?
> 
> Well it's better to have hope than to not have hope.

You have a point. But i don't think Xiaomi (or any OEM for that matter) will even consider releasing their proprietary stuff as OpenSource.
Otherwise how would stock ROMs be ANY better than custom ROMs if custom can build stuff like MiuiCamera and RealmeCamera and Fastcharge compatible kernels.

takumi021, 2022-09-12:
> As stated in the title releasing the bootloader would be very nice

Woaaahhhhhh

itsurboimasarou, 2022-09-12:
release bootloader?? really??

eclipsek20, 2022-09-12:
> release bootloader?? really??

I've just realized that I said bootloader instead of Kernal out of tiredness

Divyanshu-Modi, 2022-09-13:
They are not releasing oss kernel sources and you expected them to release the proprietary bootloader sources👍

sonic011gamer, 2022-09-13:
Well, you can always get ABL but don't expect XBL, unless it leaks somehow

closed 2024-03-25 by eclipsek20