MiCode/Xiaomi_Kernel_OpenSource #2630: Karnel sources for Redmi K50i (xagain)

DavidNingthoujam, 2022-09-21:
Please provide the karnel sources for Redmi K50i. This device has so potential chip let's take full advantage of it. Please kindly upload it so community can start improving the device!

AnyOne06, 2022-09-21:
Up for this 🙃

DavidNingthoujam, 2022-09-21:
Up

D0CT0RFATE, 2022-10-11:
@Xiaomi @Redmi Pls release the source code for Redmi K50i (XAGA) ASAP! it's been 3 months since the device has been released. Kindly help the community by giving access to the source code to the developers on order to make use of this device to it's best potential. Once again, requesting you earnestly to release kernel source for XAGA at the earliest! 

Thank you

ParampreetR, 2022-10-25:
They released kernel source for redmi note 10T pro. I think Redmi k50i will have same?
Link - https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/b17aafb30698978ff97ab9278d3e830a9a982907