MiCode/Xiaomi_Kernel_OpenSource #2643: Kernel

Githubredminote105g, 2022-09-30:
Please release kernel camellia  android S and fix android r kernel sources please

DavidNingthoujam, 2022-10-05:
Bro don't spam let's maintain simplicity. Posting once is enough, they haven't even released first ever karnel for k50i so why so hurry on getting another.

Githubredminote105g, 2022-10-10:
> Бро, не спами, давайте соблюдать простоту. Одного поста достаточно, они даже не выпустили первый в истории карнел для k50i, так зачем же

Githubredminote105g, 2022-10-10:
> Бро, не спами, давайте соблюдать просто. Одного поста достаточно, они даже не случайно первый в истории карнел для k50i, так зачем же так торопиться с получением одного еще.

Просто у нас есть ядро для андроид 11 но оно кривое а для 12 нет