MiCode/Xiaomi_Kernel_OpenSource #2649: vayu-s-oss release

acastillorobles77, 2022-10-05:
Hello, please release Poco X3 Pro Android 12 (vayu-s-oss) kernel sources, no major development can be done without android 12 kernel sources...

feiplane, 2022-10-07:
> Hello, please release Poco X3 Pro Android 12 (vayu-s-oss) kernel sources, no major development can be done without android 12 kernel sources...

U mean current one cannot work on the miui 13 + android 12 stock rom?

feiplane, 2022-10-18:
> Hello, please release Poco X3 Pro Android 12 (vayu-s-oss) kernel sources, no major development can be done without android 12 kernel sources...

Alright, u dont need newer kernel for android s, current kernel source is enough already to boot miui 13 android 12

closed 2022-10-30 by acastillorobles77