MiCode/Xiaomi_Kernel_OpenSource #2820: Fucking devices source not release

edenefri, 2022-10-21:
after 8Mth, we all been waiting dor source kernel fleur and viva,
why do make this devices suck like this with no any mod, just publish the code, we all can make some mod to keep use this brand, if still like this, some people can move to other brand, like samsung or realme etc

minionaire, 2022-10-22:
Give their apps 1 star rating on play store. They won't notice otherwise.

SamadKhafi, 2022-12-07:
Xiaomi, you need to release the kernel sources according to the license used. If not you'll be in a violation of the terms and conditions you agreed to when you used GPL licensed code.

Someone should report this license abuse to license-violation@gpl-violations.org