MiCode/Xiaomi_Kernel_OpenSource #2872: [For Fu*k sake] Release camellia-s-oss Kernel Source !!!!

willzyx-hub, 2022-12-03:
It's been a while since A12 update was released so kindly release the source !!!!

WuzwenYing, 2022-12-09:
nice try kid but the dev almost dead

willzyx-hub, 2022-12-09:
> nice try kid but the dev almost dead

Atleast i can make the twrp touch working (if possible and if i can and if source isn't broken) xD

WuzwenYing, 2022-12-09:
> > nice try kid but the dev almost dead
> 
> Atleast i can make the twrp touch working (if possible and if i can and if source isn't broken) xD

For sure tho but atleast we got a useless branch XD

willzyx-hub, 2022-12-09:
> > > nice try kid but the dev almost dead
> > 
> > 
> > Atleast i can make the twrp touch working (if possible and if i can and if source isn't broken) xD
> 
> For sure tho but atleast we got a useless branch XD

Im still helping Xiaomi to avoid getting sued by GNU 😂