MiCode/Xiaomi_Kernel_OpenSource #2970: This links are dead and i'm unable to download what is said :(

chraso, 2022-12-21:
Download Qualcomm Android Enablement Project
Acutally, this step is optional. But this project integrates many tools such as cross compiler which are required for building our kernel. Of course you can use your own tools to compile kernel, but it may be a tough work to configure.

You can download the whole QAEP on [Codeaurora](https://www.codeaurora.org/xwiki/bin/QAEP/)

$ repo init -u git://codeaurora.org/platform/manifest.git -b [branch] -m [manifest] --repo-url=git://codeaurora.org/tools/repo.git --repo-branch=caf-stable

chraso, 2022-12-24:
Please update or expla!n.

chraso, 2025-08-24:
> same. did you fixed it?

I have copied from fork's from others.

chraso, 2025-08-24:
What do you need?

Morrigan-Ship, 2025-08-25:
> ask here https://discord.com/invite/qualcommdevelopernetwork

I want ask you. you have more experience

chraso, 2025-08-31:
search for some's fork

Morrigan-Ship, 2025-08-31:
> search for some's fork

i searched not found