MiCode/Xiaomi_Kernel_OpenSource #3099: Missing dependencies and files for branch "selene-r-oss"

Ronald826, 2023-01-09:
Apparently, while compiling the _kernel source for the Redmi 10 (Selene)_, **I encountered a problem about a missing file for the Focaltech FT8719P touchscreen driver which is the FW_UPGRADE file defined in focaltech_flash.h header**. **The name of the missing file was "fw_sample.i"** **I want the branch to be fixed and merged with the required dependency or required file for a successful compilation**. For more information, please see the screenshot below:

[image]

_In addition, I am using WSL 2 on Windows 11 emulating/running Ubuntu. I was compiling the kernel source using Proton Clang and it went well until I discovered that this branch is broken_  **kindly do not ignore this problem/error**. I hope Xiaomi or this GitUser MiCode, understand.

Build Host specification:
1TB HDD, 128GB SSD, Intel Core i3-8145U, 16GB RAM, NVIDIA GeForce MX110, Windows 11, Boot Configuration with UEFI, GPT partition layout.

Branch:
**selene-r-oss**

Problem in generalized term:
**_Missing file or dependency_**

Regards,
Ronald826

Ronald826, 2023-01-09:
@mi-code Missing file on branch, selene-r-oss Redmi 10 (Selene)

Ronald826, 2023-01-09:
@mi-code @liuguo09  @ivan19871002 @thune-xiaobai @EnesSastim @xenreon, @arizaini, @XStar-Dev

Ronald826, 2023-01-10:
@mi-code @liuguo09 @ivan19871002 @thune-xiaobai @EnesSastim @XStar-Dev @arizaini @xenreon

Ronald826, 2023-01-11:
@mi-code @liuguo09 @ivan19871002 @thune-xiaobai @EnesSastim @xenreon @XStar-Dev @arizaini @xenreon

gas0324, 2023-01-12:
Thank you for your translation, but I think you should learn Chinese. As we all know, Chinese is the international standard language

CCint3, 2023-03-27:
@Ronald826, see here  https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/2602