MiCode/Xiaomi_Kernel_OpenSource #3100: I need contacts of the kernel dev of the kernel "sweet"

RUqwe, 2023-01-09:
Hello, please give me the contacts of the author of the kernel "sweet" after 2022.10. I need drivers, because the touchscreen fts_ts does not work on custom firmware, to be more precise on custom kernels.

RUqwe, 2023-01-10:
Translate with baidu 
我需要Sweet内核作者的联系人，我需要FTS_TS触摸屏驱动程序，因为触摸屏在自定义内核上不起作用。

Ronald826, 2023-01-14:
Hello @RUqwe, I recommend you base your kernel off of other sources as other sources may have that problems fixed. Try to search for your codename at Yukina-Supreme. Xiaomi is really shitty when it comes to releasing source codes.

RUqwe, 2023-01-14:
> Hello @RUqwe, I recommend you base your kernel off of other sources as other sources may have that problems fixed. Try to search for your codename at Yukina-Supreme. Xiaomi is really shitty when it comes to releasing source codes.

what is "Yukina-Supreme"?

Ronald826, 2023-01-15:
Yukina-supreme is a custom kernel. They release the devices sources too in github.

3291823, 2023-01-18:
Hi @Ronald826 , unfortunately they do not have device for sweet. Thanks for suggestion.

closed 2023-09-27 by Lsc0x80