MiCode/Xiaomi_Kernel_OpenSource #3105: Oh fuck you Xiaomi Release the fuckin camellia-s-oss

willzyx-hub, 2023-01-11:
We need this !

Ronald826, 2023-01-14:
oh my god are you serious? you can still use the old version for future android versions plus you can always patch the linux kernel code

willzyx-hub, 2023-01-15:
> oh my god are you serious? you can still use the old version for future android versions plus you can always patch the linux kernel code

If Xiaomi could release `camellia-s-oss` then it would make it easier

Ronald826, 2023-01-15:
just git fetch linux tree and git merge it. it's that simple

willzyx-hub, 2023-01-15:
> just git fetch linux tree and git merge it. it's that simple

By Linux you mean the Linux Kernel provided by Linus Torvalds ?

Ronald826, 2023-01-15:
yeah or merge the latest android kernel tree

willzyx-hub, 2023-01-15:
> yeah or merge the latest android kernel tree

so i just need to download Linux tree then copy `camellia-r-oss` sources over it ?

brian0309, 2023-03-12:
how's the progress bro? 😂😂
development of this device is dead 😂

closed 2023-03-18 by willzyx-hub

AbzRaider, 2023-05-19:
@willzyx-hub do you have any major issues on r oss just upstream it search on google there are easy guides

pankajchetry36, 2024-04-02:
There are missing drivers in old Camelia kernal source

rudy-in, 2024-04-02:
i guess i really have to work now.

pankajchetry36, 2024-04-02:
> i guess i really have to work now.

Good luck buddy

gartriss, 2024-04-03:
FIX ALREADY!! fix camellia-r-oss and release camellia-t-oss | 修复camellia-r-oss并发布camellia-t-oss!!!

rudy-in, 2024-04-04:
doing it gartriss

rudy-in, 2024-04-05:
well honestly yeah i am a bit of a new guy to the arm section would take time i am still learning how to do stuff on the android kernel out there