MiCode/Xiaomi_Kernel_OpenSource #3970: 搞快点OK？

xiaofengv233, 2023-02-19:
什么叫开源？看看隔壁op开源多积极
小米以前的发布即开源呢
大半年了连个sm8475都开不出来，隔壁都开sm8550了
kernel: https://github.com/OnePlusOSS/android_kernel_oneplus_sm8550 
kernelmodule: https://github.com/OnePlusOSS/android_kernel_modules_and_devicetree_oneplus_sm8550

liariar, 2023-02-19:
算了算了，慢慢等吧

lateautumn233, 2023-02-19:
> 什么叫开源？看看隔壁op开源多积极 小米以前的发布即开源呢 大半年了连个sm8475都开不出来，隔壁都开sm8550了 kernel: https://github.com/OnePlusOSS/android_kernel_oneplus_sm8550 kernelmodule: https://github.com/OnePlusOSS/android_kernel_modules_and_devicetree_oneplus_sm8550

复制粘贴真不错

xiaofengv233, 2023-02-19:
> > 什么叫开源？看看隔壁op开源多积极 小米以前的发布即开源呢 大半年了连个sm8475都开不出来，隔壁都开sm8550了 kernel: https://github.com/OnePlusOSS/android_kernel_oneplus_sm8550 kernelmodule: https://github.com/OnePlusOSS/android_kernel_modules_and_devicetree_oneplus_sm8550
> 
> 复制粘贴真不错

那我删了吧