MiCode/Xiaomi_Kernel_OpenSource #3974: Redmi 10c kernel source

erroor234, 2023-02-21:
Share redmi 10c kernel source, xiaomi, come to your senses

Jesuite, 2023-02-21:
Dear Xiaomi,
Kernel basically manages everything inside any software, kernel sources are usually open source by the OEM for each phone.

Now these kernel sources are needed for many modifications in the custom ROM scene, since we can create our own custom kernel by modifying it and make the ROMs smoother, we can also fix a lot of bugs in them.

AOSP (Android Open Source Project) AKA stock android doesn't run the best with a kernel which was made for MIUI or any other stock OEM ROM, and so we do some modifications to fix that and make AOSP run smoother.

Kernels also help in ROM porting development.

But why should the company care about this? Well, because it's a violation of the open-source licenses (GPL). Means that xiaomi NEEDS to release the source.

What xiaomi is doing is basically illegal, and is also a burden for the custom ROM community, it's like xiaomi doesn't care about its community.

Xiaomi please listen to us, the normal users who don't have such a reputation, so we the members of the Redmi 10C community ask that xiaomi please do something about it, as it's not a joke.

We do hope to get swift response.

liariar, 2023-02-21:
support！

ThePoisonMaster, 2023-10-05:
kernel please

Xiuzan-limit, 2024-07-05:
Still haven't got it yet, can someone help? To get it

Mistera91, 2024-07-21:
> Still haven't got it yet, can someone help? To get it

I have this phone and an outdated lineage build I found on 4pda, but no idea how they built it. I think I also found a community-made kernel once but I don’t have the link anymore