MiCode/Xiaomi_Kernel_OpenSource #3991: I suggest that you don't buy Xiaomi phones in the future.

liariar, 2023-02-24:
Isn't it good to buy a OnePlus? The configuration price is similar, but open source is much faster than Xiaomi!

mywalkb, 2023-02-24:
I totally agree with you.
I have a Redmi Note 9 Pro and I never seen kernel source. I can't fix an issue in LineageOS.
This was the first and last time.

SunixDev, 2023-02-25:
Im agree, many phone still dont have source at this time
im on a xiaomi 12t pro and no source at all

sans404-dev, 2023-02-25:
we want 12T Pro sources!!!

team-orangeBlue, 2023-02-26:
im proud of my thoughts to switch to chinese vivo phones and enjoy originos

lil-awaitz, 2023-03-06:
> 

Hello, RN9P is a unified device with RN9S, POCO M2 Pro and RN10L. these devices are running on curtana's kernel source. https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/curtana-q-oss next time check better

sleptworld, 2023-05-03:
I agreed