MiCode/Xiaomi_Kernel_OpenSource #3992: Source for XIAOMI 12t pro/k50 ultra (Dititng)

SunixDev, 2023-02-25:
Hello, please could we have kernel for xiaomi 12t pro/k50 ultra
Thanks

hesamhanari, 2023-02-26:
Hello 
I own Xiaomi 12T Pro. And it's been ~5 month but kernel source is not public
 Please releases kernel sources

Reduction77, 2023-02-27:
The same, I bought it 5 months ago but kernel source is not public, If they don't open source in two months, I'll switch to One Plus.

SunixDev, 2023-03-13:
Up, this is real k60 pro, mi 12, mi 12s, even redmi note 12 and nothink on 12t pro, for an expensive phone like this, you wont get me again xiaomi its a shame...
Not even an answer, why you make so many device to not maintain them...

cies, 2023-03-16:
There are more issues like this here: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/4024