MiCode/Xiaomi_Kernel_OpenSource #4000: K50U EEA MIUI13 相机无法连接修复包

vickwv, 2023-02-28:
- https://www.123pan.com/s/6ZTbVv-w63hH.html
- 提取码:BNhl
- 留给有需要的朋友。

vickwv, 2023-02-28:
小米不开源，国际版又故意不内置108mp驱动。所以只能自己动手解决了。

Nisekoi5, 2023-05-08:
搜了好久终于找到原因了，人脸识别出后置，微信通话只能前置，真是醉了。
大佬链接失效了可以重新发以下吗。非常感谢

vickwv, 2023-05-09:
@Nisekoi5 https://www.vickwv.world/2023/05/09/k50u-global-camera-driver/ 可以前往我的博客下载，放了链接

Nisekoi5, 2023-05-10:
> @Nisekoi5 https://www.vickwv.world/2023/05/09/k50u-global-camera-driver/ 可以前往我的博客下载，放了链接

谢谢，终于可以了。本来想着不行只能备份数据重刷EU版了，相当麻烦。头一次刷国际版不知道EEA还有这种坑。

TuFengzhi, 2024-03-20:
大佬链接失效了可以重新发以下吗。非常感谢

vickwv, 2024-03-21:
@TuFengzhi https://onedrive.live.com/?authkey=%21AHoGpES2bieClbk&id=13E7E1820DA6E4F0%21179278&cid=13E7E1820DA6E4F0

vickwv, 2024-03-21:
博客换域名了。https://www.infinitasreset.top/2023/05/09/k50u-global-camera-driver/

vickwv, 2024-03-21:
@Nisekoi5 因为这款机型国际版相机是2亿像素的

yyfmhyx, 2025-01-03:
大佬您好，博客里 OneDrive 链接失效了可以补一下么🥹

vickwv, 2025-01-04:
> 大佬您好，博客里 OneDrive 链接失效了可以补一下么🥹

https://1drv.ms/f/c/13e7e1820da6e4f0/EvDkpg2C4ecggBNOvAIAAAABOxHmCxlzMeNyAGz_l6czfw?e=hBr4HN

yyfmhyx, 2025-01-04:
@vickwv 谢谢大佬🙏