MiCode/Xiaomi_Kernel_OpenSource #4002: Xiaomi 12T pro/Redmi k50 ultra kernel source

kindle4jerry, 2023-03-01:
Please give users the freedom to use ROMs based on AOSP.
Make Xiaomi free again, please.

sans404-dev, 2023-03-02:
agree

sicksadworld44, 2023-03-03:
Total agree

liariar, 2023-03-04:
agree

SunixDev, 2023-03-13:
Up, this is real k60 pro, mi 12, mi 12s, even redmi note 12 and nothink on 12t pro, for an expensive phone like this, you wont get me again xiaomi its a shame...
Not even an answer, why you make so many device to not maintain them...

cies, 2023-03-16:
There are more issues like this here: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/4024

Theregularninja1, 2023-04-25:
Please release kernel source because I'm buying the phone soon

kindle4jerry, 2023-04-26:
> 

better not. It looks like xiaomi will not release the open source kernel. Even if Xiaomi releases the open source kernel, look the xiaomi 12 12pro 12s, it is very hard to make AOSP based ROM.

Dark1638, 2023-05-14:
Scource code pls

themagicalmammal, 2023-05-15:
Having access to the kernel source is crucial for several reasons, including:

-     **Custom ROM Development:** The release of the kernel source enables the development of custom ROMs, allowing users to enjoy the latest Android updates and additional features that may not be present in the stock firmware.

-     **Kernel Modifications:** Access to the kernel source code facilitates the ability to modify and optimize the kernel for improved performance, battery efficiency, and overall user experience.

-     **Community Collaboration:** The Xiaomi community comprises talented developers who are eager to contribute their skills and knowledge towards the improvement of the device. The availability of the kernel source fosters collaboration and innovation within the community.

byHLW, 2023-06-22:
急需

insestito, 2023-07-02:
Please

BhattJayD, 2023-09-23:
Please give us source code!!!