MiCode/Xiaomi_Kernel_OpenSource #4034: 编译小米 9Pro 5G [crux] 内核 出错，总是提示这个文件有语法错误

dajumaomao, 2023-03-19:
[image]
arch/arm64/boot/dts/qcom/pm8150.dtsi
我寻思这也没人会改这个东西啊
[image]
编译环境是 WSL Ubuntu18.04.5
P.S:
同时发现
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/crux-p-oss/drivers/input/touchscreen/fts_521/fts.c#L4658
这个地方多了一个#endif

dajumaomao, 2023-03-19:
[image]
换虚拟机也不成，彻底失望 😭😭😭😭

lateautumn233, 2023-03-19:
dts/vendor/qcom
不是dts/qcom

lateautumn
***@***.***

dajumaomao, 2023-03-19:
> dts/vendor/qcom 不是dts/qcom lateautumn ***@***.***
> […](#)

呃，我是按照wiki操作的，请问这个路径在哪里配置

dajumaomao, 2023-03-19:
[image]
用mi9  cepheus-p-oss  测试了一下 同样的错误。
烦请大佬指正。。

Gah0, 2023-03-19:
https://github.com/MiCode/kernel_devicetree/tree/elish-r-oss

dajumaomao, 2023-03-19:
> https://github.com/MiCode/kernel_devicetree

我看了这个链接，里面没有小米的 crux-p-oss，能否提供更多帮助？，谢谢！

closed 2024-03-26 by QingLea