MiCode/Xiaomi_Kernel_OpenSource #4042: Redmi 10 selene 刷入 Redmi Note 11 4G 后，WIFI 和蓝牙无法开启

CandySunPlus, 2023-03-23:
源码分支：selene-r-oss 
编译配置是用的从 官方 Redmi Note 11 4G boot.img 拆包的 config 文件
目前内核重新打包刷入后，设备能正常开机，就是 WIFI 和 蓝牙没有驱动起来。

CandySunPlus, 2023-03-23:
@ivan19871002 求大佬解答。

ivan19871002, 2023-03-24:
wifi 模块需要重新跟 内核一起编译一下

CandySunPlus, 2023-04-11:
@ivan19871002 大佬，我重新编译 MTK 相关驱动后正常了，现在有一个新的问题，boot.img 刷到 Redmi Note 11 4G  6G RAM 版本没有任何问题，但是刷到   Redmi Note 11 4G 4G RAM 版本就无法启动了，这两个版本设备还有哪些区别呢。

chikazin, 2023-04-11:
> @ivan19871002 大佬，我重新编译 MTK 相关驱动后正常了，现在有一个新的问题，boot.img 刷到 Redmi Note 11 4G 6G RAM 版本没有任何问题，但是刷到 Redmi Note 11 4G 4G RAM 版本就无法启动了，这两个版本设备还有哪些区别呢。

我也是独立编译了内核，刷入后WIFI和声音没法使用。请问你是怎么编译的驱动的呀？是把驱动源码放入内核文件夹吗还是？有相关链接嘛👀

CandySunPlus, 2023-04-12:
> 我也是独立编译了内核，刷入后WIFI和声音没法使用。请问你是怎么编译的驱动的呀？是把驱动源码放入内核文件夹吗还是？有相关链接嘛👀
https://github.com/MiCode/MTK_kernel_modules/tree/selene-r-oss
这是 MTK 相关的额外内核模块，包含 Wifi 和 蓝牙，需要和内核一起单独编译，当然你也可以把他们整合到内核里面，作为 builtin 的模块和内核编译在一起。

chikazin, 2023-04-12:
> > 我也是独立编译了内核，刷入后WIFI和声音没法使用。请问你是怎么编译的驱动的呀？是把驱动源码放入内核文件夹吗还是？有相关链接嘛👀
> 
> @chikazin https://github.com/MiCode/MTK_kernel_modules/tree/selene-r-oss 这是 MTK 相关的额外内核模块，包含 Wifi 和 蓝牙，需要和内核一起单独编译，当然你也可以把他们整合到内核里面，作为 builtin 的模块和内核编译在一起。

感谢回复🌹~想知道如何整合到内核里和内核一块编译？是否有这方面的教程链接或者commit记录👀

CandySunPlus, 2023-04-12:
@chikazin 可以看一下
https://github.com/Luquidtester/android_kernel_xiaomi_selene/commits/twelve?before=2757679fc5c2c98db45ef41c3f4d952c8f6ee155+35&branch=twelve&path%5B%5D=drivers&path%5B%5D=misc&path%5B%5D=mediatek&qualified_name=refs%2Fheads%2Ftwelve 相关的 commit

着重看 
https://github.com/Luquidtester/android_kernel_xiaomi_selene/commit/ac135e2d3d312b50b79a58a5901205e8dd9a7d35
https://github.com/Luquidtester/android_kernel_xiaomi_selene/commit/f048d1ccc7109aed44d6010031303d4f8936d1d6