MiCode/Xiaomi_Kernel_OpenSource #4044: Poco F4 GT kernel source

benzoxerxi, 2023-03-26:
Please open the source code for Poco F4 GT kernel. We need it!

yossef45601, 2023-04-08:
Please release Poco F4 gt kernel source Xiaomi please

teukuaziz010, 2023-04-08:
Please release Poco F4 gt kernel source Xiaomi please

eduardo00020, 2023-04-08:
Please release Poco F4 gt kernel source Xiaomi please🥺🥺

rimsilva, 2023-04-09:
Please open the source code for Poco F4 GT kernel. We need it!

Neo-vortex, 2023-04-23:
Comply with open-source license of Linux kernel and release the kernel code for poco x4 gt

yossef45601, 2023-04-27:
please release source code for poco f4 gt the phone has been released 1 year ago

Mahdi13864811, 2023-06-04:
Please realse source code for f4 gt I need for custom rom🙌🙏🙏🙏🙏

H4uXz, 2023-06-19:
Please release Poco F4 gt source code

hoggar11, 2023-07-04:
Please open the source code for Poco F4 GT kernel. We need it!

AngeloNF, 2023-07-24:
Please open the source code for Poco F4 GT kernel

sergio-lp, 2023-07-30:
Please release the source code for Poco F4 GT kernel.

systmrrr, 2023-08-27:
Please open the source code for Poco F4 GT kernel. We need it for custom roms. why don't you give the source code?  Will you make a statement?

babayaga2020, 2023-09-01:
Please realese the source code for POCO F4 GT KERNEL !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

IgorAzure, 2023-09-27:
Please we need the kernel source 😭

mak6021, 2023-10-09:
Let's not delay, we need the kernel source code Poco f4 gt (ingres)

AshkanRafiee, 2023-10-16:
Dear Xiaomi Kernel Open Source Team,

I hope this message finds you well. We, the Poco F4 GT community, are reaching out to request the release of the kernel source code for the Poco F4 GT device.

As enthusiastic developers and custom ROM users, having access to the kernel source code is essential for us to create and maintain custom software and optimizations for our devices. The Poco F4 GT has been available for over a year now, and we believe that the release of the kernel source code would greatly benefit the community by enabling us to enhance the functionality and performance of our devices.

We kindly urge you to consider our request and release the kernel source code for the Poco F4 GT as soon as possible. This would not only empower us to create custom ROMs but also contribute to the overall development and growth of the Poco F4 GT ecosystem.

We understand that you may have various reasons and processes in place for releasing kernel source code, but we sincerely hope that you can expedite this release to support the active and passionate Poco F4 GT community.

Thank you for your attention to our request, and we look forward to a positive response from Xiaomi. Your cooperation in releasing the source code would be greatly appreciated.

Sincerely,

aTomicTurnip, 2024-05-16:
two years already passed 😢

andrermll, 2024-09-27:
Please open the source code for Poco F4 GT kernel. We need it! 

I wait hopefully

Brubhubedits, 2025-01-16:
+1 maybe some day

hoora01, 2025-03-25:
Please open the source code for Poco F4 GT kernel

aTomicTurnip, 2025-03-25:
[image]

_*not gonna happen*_