MiCode/Xiaomi_Kernel_OpenSource #4086: ruby-s-oss

qiangua2460, 2023-04-29:
搞滴几撤，列个玩意，啷个编都编不过去，伙计你是不是把错滴放出来了？

qiangua2460, 2023-04-29:
../kernel/sched/sched.h:2365:13: 错误： inlining failed in call to always_inline ‘uclamp_se_set’: function body not available
 inline void uclamp_se_set(struct uclamp_se *uc_se,
             ^~~~~~~~~~~~~
../kernel/sched/tune.c:573:3: 附注： 从此处调用
   uclamp_se_set(&st->uclamp_req[clamp_id],
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          uclamp_none(clamp_id), false);
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../kernel/sched/tune.c:11:0:
../kernel/sched/sched.h:2365:13: 错误： inlining failed in call to always_inline ‘uclamp_se_set’: function body not available
 inline void uclamp_se_set(struct uclamp_se *uc_se,
             ^~~~~~~~~~~~~
../kernel/sched/tune.c:573:3: 附注： 从此处调用
   uclamp_se_set(&st->uclamp_req[clamp_id],
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          uclamp_none(clamp_id), false);
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../kernel/sched/tune.c: 在文件作用域：
cc1: 警告： unrecognized command line option ‘-Wno-constant-logical-operand’
make[3]: *** [../scripts/Makefile.build:334：kernel/sched/tune.o] 错误 1
make[2]: *** [../scripts/Makefile.build:637：kernel/sched] 错误 2
make[1]: *** [/home/fbl/kernel/Xiaomi_Kernel_OpenSource/Makefile:1172：kernel] 错误 2
make[1]: 离开目录“/home/fbl/kernel/Xiaomi_Kernel_OpenSource/out”
make: *** [Makefile:146：sub-make] 错误 2

qiangua2460, 2023-04-29:
In file included from ../kernel/sched/tune.c:11:0:
../kernel/sched/tune.c: 在函数‘alloc_uclamp_sched_group’中:
../kernel/sched/sched.h:2365:13: 错误： inlining failed in call to always_inline ‘uclamp_se_set’: function body not available
 inline void uclamp_se_set(struct uclamp_se *uc_se,
             ^~~~~~~~~~~~~
../kernel/sched/tune.c:573:3: 附注： 从此处调用
   uclamp_se_set(&st->uclamp_req[clamp_id],
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          uclamp_none(clamp_id), false);
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../kernel/sched/tune.c:11:0:
../kernel/sched/sched.h:2365:13: 错误： inlining failed in call to always_inline ‘uclamp_se_set’: function body not available
 inline void uclamp_se_set(struct uclamp_se *uc_se,
             ^~~~~~~~~~~~~
../kernel/sched/tune.c:573:3: 附注： 从此处调用
   uclamp_se_set(&st->uclamp_req[clamp_id],
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
          uclamp_none(clamp_id), false);
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../kernel/sched/tune.c: 在文件作用域：
cc1: 警告： unrecognized command line option ‘-Wno-constant-logical-operand’
make[3]: *** [../scripts/Makefile.build:334：kernel/sched/tune.o] 错误 1
make[2]: *** [../scripts/Makefile.build:637：kernel/sched] 错误 2
make[1]: *** [/home/fbl/kernel/Xiaomi_Kernel_OpenSource/Makefile:1172：kernel] 错误 2
make[1]: 离开目录“/home/fbl/kernel/Xiaomi_Kernel_OpenSource/out”
make: *** [Makefile:146：sub-make] 错误 2

rama982, 2023-04-30:
no issues
```
<log.txt
.....
  CC      drivers/misc/mediatek/typec/tcpc/pd_dpm_alt_mode_dc.o
  CC      drivers/misc/mediatek/thermal/common/thermal_zones/mtk_ts_imgsensor.o
  AR      drivers/misc/mediatek/typec/tcpc/built-in.a
  AR      drivers/misc/mediatek/typec/built-in.a
  AR      drivers/misc/mediatek/thermal/common/thermal_zones/built-in.a
  AR      drivers/misc/mediatek/thermal/common/built-in.a
  AR      drivers/misc/mediatek/thermal/built-in.a
  AR      drivers/misc/mediatek/built-in.a
  AR      drivers/misc/built-in.a
  AR      drivers/built-in.a
  GEN     .version
  CHK     include/generated/compile.h
  AR      built-in.a
  LTO     vmlinux.o
  MODPOST vmlinux.o
  KSYM    .tmp_kallsyms1.o
  KSYM    .tmp_kallsyms2.o
  LD      vmlinux
  SORTEX  vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm64/boot/Image
  Building modules, stage 2.
  LD [M]  drivers/video/backlight/lcd.lto.o
  LD [M]  kernel/kheaders.lto.o
  LD [M]  kernel/trace/trace_mmstat.lto.o
  LD [M]  net/ipv4/tcp_htcp.lto.o
  LD [M]  net/ipv4/tcp_westwood.lto.o
  LD [M]  sound/soc/codecs/mt6357-accdet.lto.o
  LD [M]  sound/soc/codecs/mt6359-accdet.lto.o
  MODPOST 7 modules
WARNING: sound/soc/codecs/mt6357-accdet: 'accdet_late_init' exported twice. Previous export was in vmlinux
WARNING: sound/soc/codecs/mt6357-accdet: 'accdet_read_audio_res' exported twice. Previous export was in vmlinux
WARNING: sound/soc/codecs/mt6359-accdet: 'accdet_late_init' exported twice. Previous export was in vmlinux
WARNING: sound/soc/codecs/mt6359-accdet: 'accdet_read_audio_res' exported twice. Previous export was in vmlinux
  CC      drivers/video/backlight/lcd.mod.o
  CC      kernel/kheaders.mod.o
  CC      kernel/trace/trace_mmstat.mod.o
  CC      net/ipv4/tcp_htcp.mod.o
  CC      net/ipv4/tcp_westwood.mod.o
  CC      sound/soc/codecs/mt6357-accdet.mod.o
  CC      sound/soc/codecs/mt6359-accdet.mod.o
  GZIP    arch/arm64/boot/Image.gz
  LD [M]  kernel/kheaders.ko
  LD [M]  kernel/trace/trace_mmstat.ko
  LD [M]  net/ipv4/tcp_htcp.ko
  LD [M]  drivers/video/backlight/lcd.ko
  LD [M]  sound/soc/codecs/mt6359-accdet.ko
  LD [M]  sound/soc/codecs/mt6357-accdet.ko
  LD [M]  net/ipv4/tcp_westwood.ko
+ set +x
```

qiangua2460, 2023-04-30:
> 没有问题
> 
> ```
> <log.txt
> .....
>   CC      drivers/misc/mediatek/typec/tcpc/pd_dpm_alt_mode_dc.o
>   CC      drivers/misc/mediatek/thermal/common/thermal_zones/mtk_ts_imgsensor.o
>   AR      drivers/misc/mediatek/typec/tcpc/built-in.a
>   AR      drivers/misc/mediatek/typec/built-in.a
>   AR      drivers/misc/mediatek/thermal/common/thermal_zones/built-in.a
>   AR      drivers/misc/mediatek/thermal/common/built-in.a
>   AR      drivers/misc/mediatek/thermal/built-in.a
>   AR      drivers/misc/mediatek/built-in.a
>   AR      drivers/misc/built-in.a
>   AR      drivers/built-in.a
>   GEN     .version
>   CHK     include/generated/compile.h
>   AR      built-in.a
>   LTO     vmlinux.o
>   MODPOST vmlinux.o
>   KSYM    .tmp_kallsyms1.o
>   KSYM    .tmp_kallsyms2.o
>   LD      vmlinux
>   SORTEX  vmlinux
>   SYSMAP  System.map
>   OBJCOPY arch/arm64/boot/Image
>   Building modules, stage 2.
>   LD [M]  drivers/video/backlight/lcd.lto.o
>   LD [M]  kernel/kheaders.lto.o
>   LD [M]  kernel/trace/trace_mmstat.lto.o
>   LD [M]  net/ipv4/tcp_htcp.lto.o
>   LD [M]  net/ipv4/tcp_westwood.lto.o
>   LD [M]  sound/soc/codecs/mt6357-accdet.lto.o
>   LD [M]  sound/soc/codecs/mt6359-accdet.lto.o
>   MODPOST 7 modules
> WARNING: sound/soc/codecs/mt6357-accdet: 'accdet_late_init' exported twice. Previous export was in vmlinux
> WARNING: sound/soc/codecs/mt6357-accdet: 'accdet_read_audio_res' exported twice. Previous export was in vmlinux
> WARNING: sound/soc/codecs/mt6359-accdet: 'accdet_late_init' exported twice. Previous export was in vmlinux
> WARNING: sound/soc/codecs/mt6359-accdet: 'accdet_read_audio_res' exported twice. Previous export was in vmlinux
>   CC      drivers/video/backlight/lcd.mod.o
>   CC      kernel/kheaders.mod.o
>   CC      kernel/trace/trace_mmstat.mod.o
>   CC      net/ipv4/tcp_htcp.mod.o
>   CC      net/ipv4/tcp_westwood.mod.o
>   CC      sound/soc/codecs/mt6357-accdet.mod.o
>   CC      sound/soc/codecs/mt6359-accdet.mod.o
>   GZIP    arch/arm64/boot/Image.gz
>   LD [M]  kernel/kheaders.ko
>   LD [M]  kernel/trace/trace_mmstat.ko
>   LD [M]  net/ipv4/tcp_htcp.ko
>   LD [M]  drivers/video/backlight/lcd.ko
>   LD [M]  sound/soc/codecs/mt6359-accdet.ko
>   LD [M]  sound/soc/codecs/mt6357-accdet.ko
>   LD [M]  net/ipv4/tcp_westwood.ko
> + set +x
> ```
How do you compile? Can you tell me the compiling environment and tools?

qiangua2460, 2023-05-03:
缺imx766ofilm2_mipi_raw文件夹

qiangua2460, 2023-05-03:
缺imx766ofilm2_mipi_raw文件夹，麻烦补齐

qiangua2460, 2023-05-03:
[image]

qiangua2460, 2023-05-03:
[image]

qiangua2460, 2023-05-03:
[image]

qiangua2460, 2023-05-06:
请把缺少的文件夹及文件补充上来

closed 2024-12-12 by qiangua2460