MiCode/Xiaomi_Kernel_OpenSource #4092: Xiaomi violates GPL licenses

sleptworld, 2023-05-03:
**Kernel Sources Unreleased Models:**
- Redmi Note 11 / NFC (spes/spesn)
- Redmi Note 11 Pro 5G / Redmi Note 11E Pro (veux)
- Xiaomi 12T (plato)
- Xiaomi 12T Pro / Redmi K50 Ultra (diting)
- Redmi Note 12 4G (tapas/topaz)
- Redmi Note 12 5G (sunstone)
- Redmi Note 12 Pro 4G (sweet_k6a)
- Redmi Note 12S (sea)
- Redmi K60 / POCO F5 Pro (mondrian)
- Redmi K60E (rembrandt)
- Xiaomi 12 Pro Dimensity Edition (daumier)

satishs22, 2023-05-05:
https://xiaomifirmwareupdater.com/

can try this

cuynu, 2023-05-06:
K60 https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/mondrian-s-oss

cuynu, 2023-05-06:
Xiaomi 12 Pro
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/daumier-s-oss

sleptworld, 2023-05-07:
> Xiaomi 12 Pro https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/daumier-s-oss

Ok,  i will remove these items from the list.

tptnam, 2023-05-07:
and 11t (agate)

popy2k14, 2023-05-10:
+1 for 11T (agate/amber) @mi-code

ech0devv, 2023-05-11:
@mi-code 12t

Sushrut1101, 2023-05-12:
+ Redmi 10A
+ viva
+ evergo-s-oss

jstsdev, 2023-05-19:
i need Redmi Note 12 4G (tapas/topaz) please

lvctr, 2023-05-20:
One more for the list:

- Redmi Note 12 Turbo / Poco F5 (marble)

EvoTeamMalaysia, 2023-05-22:
11T Agate not been release till now

str4fe, 2023-05-27:
spes/spesn now is released

TalasDigital, 2023-05-29:
12S Ultra sources are still missing after only 10months. Is there any hope?

AnyOne06, 2023-07-12:
Diting released.

eclipsek20, 2023-08-26:
Xiaomi 12 still missing (cupid)

popy2k14, 2023-08-26:
It's a shame that we have to upgrade because of missing source code. Those mobiles are good to go a few years.

Xiaomi, come on.

eclipsek20, 2023-08-26:
> It's a shame that we have to upgrade because of missing source code. Those mobiles are good to go a few years.
> 
> Xiaomi, come on.

Just buy a pixel and don't deal with this shit

popy2k14, 2023-08-26:
@eclipsek20 That's one reason not too buy an xiaomi again!

AnyOne06, 2023-08-26:
Xiaomi best

eclipsek20, 2023-08-26:
> @eclipsek20 That's one reason not too buy an xiaomi again!

Yeah I'll go with a pixel or nothing

cuynu, 2023-08-27:
Not just xiaomi, most manufacturer from china are the same, like Lenovo and ODMs like SDMC, QIN, HISENSE, they are violates GPL critical and NEVER release kernel source. China is the most country that violations almost any GPL rules, they only only only want CLOSED SOURCE

newlund, 2023-11-13:
The following kernel source is incomplete. Should be added to the list again.

Redmi K60 / POCO F5 Pro (mondrian)

**Xiaomi please don't violate the GPL and support the community. Show that you are not a parasite company.**

blacklightpy, 2024-08-11:
Redmi Note 12S missing main camera drivers

popy2k14, 2024-08-11:
I bought an Pixel 8 Pro!
Never ever Xiaomi Smartphones!

gartriss, 2024-08-29:
Redmi Note 10 5G (camellia), incomplete and broken kernel source.

madurapa, 2024-10-19:
Still not available for Redmi 10A

wahyuief, 2024-12-03:
Redmi K50 Gaming/Poco F4 GT (ingres) too

Become-ILLUSORY, 2024-12-22:
redmi k50 pro still missing

SyedHasan110, 2025-06-04:
The only solution for this problem is everybody stop buying xiaomi phones.

SyedHasan110, 2025-06-04:
Yes, Exactly they earn money using ads but at the cost of destroying user UI/UX Experience of HyperOS. Plus Integrating bloatware shit this makes user annoyed and the user turn towards custom rom for bloatware free and clean UI/UX. Current, I have Installed HyperOS Indonesin version and there are two messaging apps and several unwanted apps that i do not use.

madurapa, 2025-06-05:
Based on my research, Nokia or Motorola are great options if you're on a budget. 

For flagship devices, I’d recommend Nothing or Pixel. 

I’ve used Nokia 6, Pixel 3, and Pixel 5, all fantastic phones that I absolutely loved!

`Though long discontinued, my all-time favorite remains the YotaPhone, a true pioneer with its dual-screen design. RIP to an underrated legend!`

Zaptyp, 2025-09-13:
Share cupid kernel sources if is near a end of life for getting major android updates!