MiCode/Xiaomi_Kernel_OpenSource #4190: Dear Xiaomi

Vijayraobappu, 2023-05-25:
Kernel basically manages everything inside any software, kernel sources are usually open source by the OEM for each phone.

Now these kernel sources are needed for many modifications in the custom ROM scene, since we can create our own custom kernel by modifying it and make the ROMs smoother, we can also fix a lot of bugs in them.

AOSP (Android Open Source Project) AKA stock android doesn't run the best with a kernel which was made for MIUI or any other stock OEM ROM, and so we do some modifications to fix that and make AOSP run smoother.

Kernels also help in ROM porting development.

But why should the company care about this? Well, because it's a violation of the open-source licenses (GPL). Means that xiaomi NEEDS to release the source.

What xiaomi is doing is basically illegal, and is also a burden for the custom ROM community, it's like xiaomi doesn't care about its community.

Xiaomi please listen to us, the normal users who don't have such a reputation, so we the members of the Redmi note 10s community ask that xiaomi please do something about it, as it's not a joke.

We do hope to get swift response.

romman999, 2023-05-26:
redmi note 10s rosemary already available,,🙄

Vijayraobappu, 2023-06-01:
> redmi note 10s rosemary already available,,🙄

Not available boi