MiCode/Xiaomi_Kernel_OpenSource #4236: Errors while compiling spes-r-oss branch

PavelAtr, 2023-06-10:
Please help me get rid of compilation errors at the final stage of spes-r-oss branch.

``
WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
drivers/gpu/drm/drm_sysfs.o: In function `doze_backlight_store':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/gpu/drm/drm_sysfs.c:350: undefined reference to `dsi_bridge_disp_set_doze_backlight'
drivers/gpu/drm/drm_sysfs.o: In function `doze_backlight_show':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/gpu/drm/drm_sysfs.c:359: undefined reference to `dsi_bridge_disp_get_doze_backlight'
drivers/gpu/drm/drm_sysfs.o: In function `mipi_reg_store':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/gpu/drm/drm_sysfs.c:318: undefined reference to `dsi_display_mipi_reg_write'
drivers/gpu/drm/drm_sysfs.o: In function `mipi_reg_show':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/gpu/drm/drm_sysfs.c:310: undefined reference to `dsi_display_mipi_reg_read'
drivers/input/touchscreen/nt36525b_spi/nt36xxx.o: In function `lct_nvt_tp_gesture_callback':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/input/touchscreen/nt36525b_spi/nt36xxx.c:2918: undefined reference to `set_lcd_reset_gpio_keep_high'
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/input/touchscreen/nt36525b_spi/nt36xxx.c:2912: undefined reference to `set_lcd_reset_gpio_keep_high'
drivers/thermal/thermal_core.o: In function `thermal_exit':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/thermal/thermal_core.c:1997: undefined reference to `drm_unregister_client'
drivers/thermal/thermal_core.o: In function `thermal_init':
/home/pavel/xiaomi-spesn/kernels/linux-source-4.19.157-tiny/drivers/thermal/thermal_core.c:1972: undefined reference to `drm_register_client'
``

PavelAtr, 2023-06-11:
/techpack is empty in branch spes-r-oss

closed 2023-06-11 by PavelAtr

Guilher89, 2023-08-20:
where did you find the build guide for spes-r-oss? i need it, please help me

PavelAtr, 2023-08-20:
I copy /techpack directory from
https://github.com/PixelExperience-Devices/kernel_xiaomi_sm6225.git
repository. And then build kernel.

Guilher89, 2023-08-21:
thank you so much, i'm using pixel experience too

Guilher89, 2023-08-23:
did you manage to compile?, I need your knowledge could you tell me which configuration you used and which tool you used?

PavelAtr, 2023-08-23:
Email me pvsamsonov76@gmail.com