MiCode/Xiaomi_Kernel_OpenSource #4238: /techpack is empty in branch spes-r-oss

PavelAtr, 2023-06-11:
/techpack is empty in branch spes-r-oss !

Guilher89, 2023-08-23:
did you manage to compile?, I need your knowledge could you tell me which configuration you used and which tool you used?

Guilher89, 2023-09-27:
I hope this message finds you well.  I'm contacting you for information and guidance on compiling the SPES kernel.  I am interested in understanding the compilation process and its specificities.

 So that I can gain a more comprehensive understanding, I would like to ask you a few specific questions:

 1. **Toolchain Used:** Which toolchain (set of compilation tools) did you use to compile the SPES kernel?  Was it a specific toolchain recommended by the manufacturer or one of your choice?

 2. **Compilation Script:** During the compilation process, did you use any custom kernel compilation script or a manufacturer-supplied script?  If yes, could you share details about this script or its settings?

 3. **Specific Settings:** Were there any special settings or adjustments you needed to make when compiling the SPES kernel, especially related to specific hardware or optimizations?

 4. **Issues or Challenges:** While compiling the kernel, did you face any specific issues or challenges that you would like to highlight?  How did you resolve these issues, if any?

 If you are willing to share your experiences and knowledge about compiling the SPES kernel, I would be extremely grateful.  Your information will be invaluable to my own compiling journey.

 Please know that any details or guidance you can provide will be highly appreciated.  If you prefer to share this information in a more private setting, I am happy to exchange private messages or emails, whichever is most convenient for you.

 Thank you in advance for your help and collaboration.  Your knowledge will be of great benefit.