MiCode/Xiaomi_Kernel_OpenSource #4246: Redmi 2A (lte26007) Source code

sunshijiang, 2023-06-14:
Dear Xiaomi,

I am writing to request a copy of the GPL kernel source code.

I understand that the GPL kernel is licensed under the GNU General Public License, which requires that any modifications to the kernel be made available under the same license. I am willing to comply with this requirement.

I would be grateful if you could send me a copy of the GPL kernel source code as soon as possible.

Thank you for your time and attention to this matter.

亲爱的小米，

我写信是为了索取 GPL 内核源代码的副本。

我了解 GPL 内核是根据 GNU 通用公共许可证获得许可的，该许可证要求对内核的任何修改都在同一许可证下提供。我愿意遵守这个要求。

如果您能尽快给我一份 GPL 内核源代码，我将不胜感激。

感谢您花时间关注此事。

Alpa7000, 2023-06-14:
这是联芯的吧，可能内核都不完整

sunshijiang, 2023-06-14:
> 这是联芯的吧，可能内核都不完整

我看github上面有一个[nichcream/kernel_xiaomi_lte26007](https://github.com/nichcream/kernel_xiaomi_lte26007)，不知道这个和小米官方的有什么区别。