MiCode/Xiaomi_Kernel_OpenSource #4682: [Request] Xiaomi Redmi 12C Global (earth)

buadambabaisi, 2023-06-21:
Please release the kernel for Redmi 12C (earth)

Woomymy, 2023-06-21:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/earth-s-oss

lls2wow, 2024-01-07:
For Redmi12C what config should I choose for ?  from Xiaomi_Kernel_OpenSource_12C/arch/arm64/configs.

lls2wow, 2024-01-07:
@Woomymy thank you for kindly response

Woomymy, 2024-01-07:
> For Redmi12C what config should I choose for ?  from Xiaomi_Kernel_OpenSource_12C/arch/arm64/configs.

You should probably use earth_defconfig

lls2wow, 2024-01-07:
```Can't find default configuration "arch/arm64/configs/earth_defconfig"!
***
../scripts/kconfig/Makefile:112: recipe for target 'earth_defconfig' failed
make[2]: *** [earth_defconfig] Error 1
/Xiaomi_Kernel_OpenSource_12C/Makefile:586: recipe for target 'earth_defconfig' failed
make[1]: *** [earth_defconfig] Error 2
make[1]: Leaving directory '/Xiaomi_Kernel_OpenSource_12C/out'
Makefile:152: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2```

Sorry I did not find the earth_defconfig in the path Xiaomi_Kernel_OpenSource_12C/arch/arm64/configs. Could you help me check what the right config?

lls2wow, 2024-01-08:
@mi-code 你可以告诉我redmi 12c应该使用代码库中的哪个配置来编译吗

lls2wow, 2024-01-08:
@Woomymy 解决了 原来是我git clone的时候 -b参数没生效 所以没这个配置 重新弄了就好了 郁闷死了

Reyhank45, 2024-06-16:
Help im getting this error while compiling  :
[image]

oceanzhang01, 2024-12-18:
> @Woomymy 解决了 原来是我git clone的时候 -b参数没生效 所以没这个配置 重新弄了就好了 郁闷死了

兄弟 你编译成功后能刷进去开机吗 ？

SpeechlessMatt, 2025-02-16:
> > [@Woomymy](https://github.com/Woomymy) 解决了 原来是我git clone的时候 -b参数没生效 所以没这个配置 重新弄了就好了 郁闷死了
> 
> 兄弟 你编译成功后能刷进去开机吗 ？

我的也开不了机 用了LineageOS的试过也开不了机 破防了

SpeechlessMatt, 2025-02-17:
> > > [@Woomymy](https://github.com/Woomymy) 解决了 原来是我git clone的时候 -b参数没生效 所以没这个配置 重新弄了就好了 郁闷死了
> > 
> > 兄弟 你编译成功后能刷进去开机吗 ？
> 
> 我的也开不了机 用了LineageOS的试过也开不了机 破防了

Clang的问题，我换了clang编译成功了，用的lineage的rebase,官方的有问题，而且不是最新的kernel

Oscar1640, 2025-04-04:
> Problema de ruido metálico. Cambié clang y compilé exitosamente. Utilicé la rebase de linaje. El oficial tiene problemas y no es el último kernel.

@SpeechlessMatt 你是怎么做到的，如何让内核工作？

Oscar1640, 2025-04-05:
> Clang的问题，我换了clang编译成功了，用的lineage的rebase,官方的有问题，而且不是最新的kernel

@SpeechlessMatt  你是怎么做到的，如何让内核工作？

SpeechlessMatt, 2025-04-05:
> > Clang的问题，我换了clang编译成功了，用的lineage的rebase,官方的有问题，而且不是最新的kernel
> 
> @SpeechlessMatt  你是怎么做到的，如何让内核工作？

请试用第三方的clang和内核仓库，官方的似乎有缺失